FastLED/FastLED #373: SK6812 support

eltang, 2016-11-19:
Would it be possible for this chipset to be supported by FastLED?

kriegsman, 2016-11-19:
Yep - it already works.   It's a clone of the WS2812, so you can just tell FastLED that you're using WS2812, and the SK6812's will "just work".

closed 2016-11-19 by kriegsman

eltang, 2016-11-19:
Is the information on [this page](https://learn.adafruit.com/adafruit-neopixel-uberguide/advanced-coding) inaccurate then?

focalintent, 2016-11-19:
There are two flavors ofnsk6812 - RGB And RGBW - fastled doesn't supportnRGBW yet (and there's already an open issue for tracking that)

eltang, 2016-11-19:
Oops, I hadn't noticed that issue. Has any progress been made since June?

eltang, 2016-11-19:
For your information, I'm going to be getting some SK6812RGBW LEDs along with a Teensy 2.0 soon.  If RGBW LED support is something you're actively working on, I'd be happy to test it for you.

kriegsman, 2016-11-19:
Thank you.  Sadly, there's a been a death in our family, and we're attending to that first and foremost.  But thank you for the offer of help; we'll come back around to that.

eltang, 2016-11-19:
Sorry for your loss. I may need to have the bare AVR support we discussed in the other issue before I start testing, though, as I do not want my project to be dependent on the Arduino IDE.

eltang, 2017-01-08:
I have my hardware now. Just let me know when you'd like me to test something.

Ethanmose, 2017-07-06:
sk6812 have sk6812 rgbw,sk6812 rgb,sk6812 wwa,sk6812 3535 ,sk6812 white,ws2812b also have ws2812b rgb,ws2812b 3535. and sk6812 is better than ws2812b,tansmission speed more faster.

Blink515, 2017-10-11:
> If you just ignore the white led, will it work as of now?

No it will not work. The colors will not be what you expect.

DarkiStar, 2017-10-15:
Here 
[Link to FastLED with RGBW](http://www.partsnotincluded.com/neopixels/fastled-rgbw-neopixels-sk6812/)
someone found a workaround for using rgbw with FastLed.

TheCellMC, 2017-11-18:
any news on this?

coryking, 2017-12-12:
Just a note that I've added RGBW support to [my fork of FastLED](/coryking/FastLED) -  Works only on ESP8266.  So far, pretty sweet!