FastLED/FastLED #374: WS2813 pixels show random colours

cosmicnut, 2016-11-20:
I've got a string of 144 WS2813 pixels I'm driving with an arduino Duo. 
Any example that can use the whole string gives random results. Flashing pixels of all colours but with not decernable pattern
The RGBCalibrate example gives a random block of white pixels chases up and down the string
Blink, like any other single LED sketch I've tried gives nothing at all.

I've tried all the WS281x chipsets adn the neopixel one and nothing changes

I'm powering via a 12v supply capable of 6amp that I'm dropping to 5v (5.3v) via a buck module capable of 6amp. 
I can get the whole string to light but I can't control the colour

KeirSweeney, 2016-12-21:
Hi @cosmicnut, I'm also having the same issue with WS2813 LED's where the data line is causing the lights to flash uncontrollably. If I plug the LED power cables directly into the Arduino the lights are stable, however from an external power supply the lights don't behave how I would expect.

Did you make any progress in fixing this?

Daniel-dk, 2016-12-21:
Are the Grounds conencted between the power supplies, LEDs and the Arduino ?
the Due 3v3 might also be a little low for driving the WS LED data lines ( if there is a little noise the data signal is lost )

KeirSweeney, 2016-12-21:
Currently no, I'm unsure on how to ground the Arduino into the PSU and worried about frying it. Do I just need to use the Ground port from the Arduino MEGA and connect that to what part of the PSU?

KeirSweeney, 2016-12-21:
@Daniel-dk Fixed it! Sharing the ground from the LED into the Arduino did the trick.
Thanks for replying so fast!

cosmicnut, 2016-12-22:
D'oh! Yep, that did it for me as well.

Daniel-dk, 2016-12-23:
Sharing the grounds shouldn't be an issue for most cases ( its a common reference point for all devices  )

phugoid7, 2016-12-23:
How are you guy's using the double input signal feature of the WS2813? Are you using two input wires from the arduino to the strip or the same input wire from the arduino split over the two signal ports on the strip?
thanks

KeirSweeney, 2016-12-23:
I'm currently using the WS2812 setting on the Arduino using the Fast Led library. My RGB wasn't in the same order so I had to change that.

For the data I split the data signal from the Arduino into both data lines on the LED strip. It will work with just one connected but I've got them both hooked up for now! Works perfectly

phugoid7, 2016-12-23:
Great, thanks. Do you think the data break feature will still work with only one data connection pin connected to the arduino or are you just connecting 2 for good measure?

KeirSweeney, 2016-12-23:
I think theoretically it should. But I have tested it myself yet, after Christmas I will be cutting a strip up so I can test it then by soldering only one data line together, splitting the data signal could lead to loss of data over a long strip, so I'm trying to find the best solution!

witop, 2017-04-10:
we supply the ws2813 Addressable led strip. 30led &60Led&144led. 2 years warranty and ce rohs certification. sample is ok. learn more please contact jessie@witop-tech.com

alvin666666, 2017-04-20:
hello,we are the led strip custom factory,new product cs2803, also,breakpoint continue 5v addressable,and 12v addresssable breakpoint continue ws2815.   If you want toi know more,send email to me,i can send datasheet to you.Email:Alvin@witop-tech.com

vshymanskyy, 2017-05-04:
I had no issues with WS2813B (it works), but my strip has 2 data lines for better reliability (DATA and DATA_AUX).
For now it looks like DATA_AUX is not used. Is it supported by FastLED?

Ethanmose, 2017-07-06:
as you know,ws2813 have the function of break-point resume,so,it is have 4 line,DI BI line,but ws2812b only have DI ,not BI,so,maybe you can try do not connect the BI to the arduino,if connect ,it will affect the signal ,i think. I think ws2813 led strip is betetr than ws2812b or sk6812.

jvanderberg, 2017-10-07:
I had the same issue.  When I connected the grounds of the Arduino's power supply and the LED strips power supply, the issue when away.

VinceW31, 2017-10-31:
For Ws2813 Both Data lines should be connected together to the same data pin on the Arduino etc. Ws2813 strip will work ok if only the DI line is connected and BI left disconnected, but in this config the first LED will not work as the chip expects the same data on both inputs. After each LED chip the BI line from the input of the first chip is directly connected to the DI input of the next chip (which is also connected to the D0 of the first chip. Therefore only one Data pin is needed to feed both inputs and is connected to both DI and BI inputs.

closed 2019-01-20 by focalintent

daddio5023, 2020-01-14:
I have a WS2811 12V strip with 97 pixels on it connected to an arduino uno.  I'm trying the blink demo for FastLED that is set up for only 20 pixels and they all show random colours (it doesn't matter how many pixels I set it up for.  When I connect the arduino power ground to the 12 V power supply ground the colour stabilizes and works but only the first light blinks.  Any suggestions?

marmilicious, 2020-01-14:
A common ground is always required between the LEDs and the controller and the power supply, otherwise you will get random blinking as you did.

The blink example only blinks the first pixel, leds[0], so what you are seeing is correct (once you added the common ground wire).

For general questions like these, please check out the FastLED Reddit group: https://www.reddit.com/r/FastLED/