FastLED/FastLED #376: Is FastLed bound to Arduino environment?

zallesov, 2016-11-24:
FastLed is an essential part of my application. But I'm evaluating a possibilitiy to migrate from Arduino to Atmel Software Framework and FreeRTOS for the ARM/SAM/DUE hardware.

Can FastLed be used in context of Atmel Software Framework with no "#include <Arduino.h>"? 
http://www.atmel.com/tools/avrsoftwareframework.aspx

Has it been ever cheked within the RTOS apps for the time sensitive Leds like ws2812?
http://www.freertos.org/about-RTOS.html

Thax for the answers even if they are negtive.

jrobeson, 2016-11-24:
read up here https://github.com/FastLED/FastLED/issues/220

focalintent, 2016-11-25:
I haven't done any work with getting fastled to run in an RTOS environment.  It might, it might not ¯\_(ツ)_/¯

zallesov, 2016-11-26:
Thanx for the quick answers.
So As far as I can conclude the answers are. 
 - FastLED can potentially do without Arduino definitions, but it seems like there is no ready to adopt example.
- It is uncertain whether It would work within RTOS. For me, it seems like it really depends on the number of LEDs and other tasks performed on the chip. The right way to drive time-sensitive leds with the RTOS would be to use a separate controller with DMA like described here https://mcuoneclipse.com/2014/11/10/neoshield-ws2812-rgb-led-shield-with-dma-and-nrf24l01/

focalintent, 2019-01-20:
FastLED will become less Arduino dependent in the future (there will be a null platform added for building for non-arduino/mcu style hardware - and as part of the RGBW/16-bit re-write I will try to peel out any remaining Arduino dependencies (or make it possible to define them out).

closed 2019-01-20 by focalintent