FastLED/FastLED #377: DemoReel100 Example: Rainbow / Base colour not being cycled in my project

NonDualChris, 2016-11-28:
Hi (Forgive my inexperience here).

I am using the example code with minimal alterations (only DATA_PIN, NUM_LEDS, LED_TYPE and FRAMESPERSECOND).
I am using an array of 60 WS2812B LEDS and an Arduino Nano controller.
The Rainbow pattern is a static rainbow. There is no cycling base colour.
I see how it's meant to work with the gHue value. Sadly, it's not happening.

I'm happy to try a few things to get this working if it will assist the project.
Any thoughts?
Chris

NonDualChris, 2016-11-28:
In addition - Your 'rainbow_march' example works fine for me - The colour cycling is fine.
This is where the Hue value is incremented within the function itself, rather than by time delay in the main loop.

marmilicious, 2016-11-29:
@NonDualChris Hi there. Github here is only used for library bug reports.  For general and code related assistance post you question to the FastLED G+ group.
http://fastled.io/+

Share your code on Pastebin or gist and include a link to it in your post.  Also good to share what hardware you're using, and what version of the Arduino IDE and FastLED library you're using just to be as clear as possible.

kcouck, 2017-01-11:
I've had the same issue with static variables not incrementing when being passed into a function call. When the variable is only used locally (e.g. in the 'loop()' function, it works fine...

kriegsman, 2017-01-11:
@kcouck: C and C++ pass arguments by _value_, not by _reference_.   So when you call a function using a static variable as a parameter, you're just passing in the _current value_ of the static.  Inside the function, that value is bound to a new local variable (the named parameter).  Altering the local parameter won't have any effect on the 'outside' static.    Google "passing static variables as arguments" and you should find some good info.  It's not specific to FastLED, or Arduino.  It's just how the C/C++ language works.  (And yes, there are ways to do what you probably want; Google around for 'passing _references_ to static variables', perhaps?)

kcouck, 2017-01-12:
I guess my remark wasn't clear :/
The issue I'm having is the static variable being 'stuck' on the same value in the calling loop where the increment happens as well, when it is being passed as a parameter to a function. Putting the function code into the loop itself works... I'll try to fix up some example code tomorrow... It's really weird, I need to cross-check the IDE and Hardware (this happens on VS2015 with VisualMicro and a Teensy 3.2 - should try with default Arduino IDE and a 'real' Arduino)

kcouck, 2017-01-12:
As a sidenote (sorry for the spam), it's still not a specific FastLed issue...

kcouck, 2017-01-15:
Forget my remark; just retested this and now it works fine; must have been a bug in Visual Micro or the Arduino IDE...
Working fine with Arduino 1.8.0 now...

closed 2019-01-20 by focalintent