FastLED/FastLED #378: Loading very large palettes will fail due to 8-bit counter

jake-b, 2016-12-07:
The =operator and loadDynamicGradientPalette for CRGBPaletteXX will first scan a palette to determine its length, looking for a 255 "end of gradient" marker.

Very large gradients that are larger than the 8-bit counter will fail to find this, and the 8-bit counter will wrap.  This causes an infinite loop and WDT reset.

By using a 16-bit integer for the counter, larger palettes can be loaded properly.  Granted, they're truncated by the 16-entry limit, but this is better than an unexplained infinite loop.

kriegsman, 2016-12-07:
Thanks for the catch!

jake-b, 2016-12-07:
Trivial pull request here: https://github.com/FastLED/FastLED/pull/379

kriegsman, 2016-12-09:
Thanks for the catch, and the PR!  Thoughtful help like this is always appreciated.

closed 2016-12-09 by kriegsman