FastLED/FastLED #380: ColorFromPalette loops back to first color with index > 240

clinder, 2016-12-09:
When calling ColorFromPalette with a CRGBPalette16 and a heat index > 240, I expect the function to return a color that is interpolated between the second to last color and the last color in the palette. Instead it seems to interpolate between the second to last color and the first color in the palette. In the code below I would expect the LED to fade from cyan to blue to red but instead it fades from cyan to blue to white.

#include <FastLED.h>

#define NUM_LEDS 1
#define DATA_PIN 0

CRGB leds[NUM_LEDS];

DEFINE_GRADIENT_PALETTE( heatmap_gp ) {
  0,     255,  255,  255,
128,   255,0,0,
224,   0,255,0,
240,   0,0,255,
255,   255,0,0 };

CRGBPalette16 myPal = heatmap_gp;

void setup() {
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
}

void loop() {
  for(int i = 235; i < 255; ++i) {
    leds[0] = ColorFromPalette( myPal, i);
    FastLED.show();
    delay(1000);
  }
}

It looks like the issue is related to how the function:
CRGB ColorFromPalette( const CRGBPalette16& pal, uint8_t index, uint8_t brightness, TBlendType blendType)
deals with hi5 indexing the last chunk:
        if( hi4 == 15 ) {
            entry = &(pal[0]);

Is this the intended behavior? I realize I can work around it by just ignoring the last palette entry and never indexing above 240 but it would be preferable to use the whole range.

Thanks.

marmilicious, 2016-12-09:
@clinder, I did see this in the docs:
"Recommend that you use values 0-240 rather than the usual 0-255, as the last 15 colors will be 'wrapping around' from the hot end to the cold end, which looks wrong."
from here:
http://fastled.io/docs/3.1/group___colorpalletes.html

kriegsman, 2016-12-09:
Yep. In order to support smooth blending for "circular" palettes, each color blends with "the next"-- and in the case of the last color, "the next" is the first color again since it wraps around. 

If you want to avoid wrapping, just do this:

CRGB c = ColorFromPalette( myPal, scale8( i , 240 ) );

This is what I do. It lets me program with 0-255 for i, and since scale8 is extremely small and fast the code overhead is basically unnoticeable. 

The "circular wraparound" of palette colors is by design, and since there's an easy way for programmers to avoid if if they wish, I think we're likely to leave it as is. 

Thanks for the feedback, and always happy to see people using palettes (and FastLED)!

closed 2016-12-09 by kriegsman