FastLED/FastLED #386: Compiling on OSX for ESP8266 does not work

jeroenh, 2016-12-15:
I am working on a collaborative project using ESP8266 with neopixel leds (to make a word-clock). We started out with the Adafruit library, but now someone has contributed code to run with the FastLED library.
The code seems to compile for almost everyone, but not for me, running on OSX.

I'm running Arduino 1.6.13, have installed FastLED 3.1.3 and I get the following messages when I try to compile my project. It does not matter which ESP board I try to compile for, in this case I try to use NodeMCU 0.9:

```
Arduino: 1.6.13 (Mac OS X), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from /Users/jeroen/Projects/wordclock/fastled-wordclock/fastled-wordclock.ino:3:0:
/Users/jeroen/Documents/Arduino/libraries/FastLED-3.1.3/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /Users/jeroen/Documents/Arduino/libraries/FastLED-3.1.3/FastLED.h:65:0,
                 from /Users/jeroen/Projects/wordclock/fastled-wordclock/fastled-wordclock.ino:3:
/Users/jeroen/Documents/Arduino/libraries/FastLED-3.1.3/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
/Users/jeroen/Projects/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: fastled-wordclock.ino.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

focalintent, 2016-12-15:
You don't mention what version of esp8266-arruino you are using.  

Also - search for "section `.text' will not fit in region `iram1_0_seg'" on google for discussions on how to fix this.

jeroenh, 2016-12-15:
I'm using the git stable checkout of esp8266-arduino which is last updated on Nov 14.
https://github.com/esp8266/Arduino

I have googled for the error message, but the frustrating thing seems to be that the exact same code and library versions compile on Linux, but do not compile on OS X.
So far all changes that I've seen suggest making changes somewhere in the code or in one of the libraries, which does not seem to be the problem here.

focalintent, 2016-12-15:
Unfortunately I'm taking a break from active work on the library due to a death in my household - the g+ group (http://fastled.io/+) may have folks who can help work around this.  Also at least one of the matches for this error indicated that it happened more often on the mac which sounds like an issue with the platform tools built for the mac?

jeroenh, 2016-12-15:
Fixed it.

It turns out that apparently there is a difference between installing via git in `Documents/Arduino/hardware/esp8266com/` and installing through the board manager (after adding the URL: http://arduino.esp8266.com/stable/package_esp8266com_index.json) 
Using the latter method fixed the problem.

closed 2016-12-15 by jeroenh