FastLED/FastLED #390: Error in line 68 "no matching function for call to 'random()'

christherock23, 2016-12-24:
Hey guys having a bit of an interesting issue here, when i try to compile the fire2012 with palette i get an error on line 68 - 

random16_add_entropy( random());

this is the error - 

**************************************************************************************************************
Arduino: 1.6.13 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 512000, 4M (3M SPIFFS)"

In file included from C:\Users\Chris\Documents\Arduino\libraries\FastLED\examples\Fire2012WithPalette\Fire2012WithPalette.ino:1:0:

C:\Users\Chris\Documents\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003

 #    pragma message "FastLED version 3.001.003"

                     ^

In file included from C:\Users\Chris\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\Chris\Documents\Arduino\libraries\FastLED\examples\Fire2012WithPalette\Fire2012WithPalette.ino:1:

C:\Users\Chris\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Users\Chris\Documents\Arduino\libraries\FastLED\examples\Fire2012WithPalette\Fire2012WithPalette.ino: In function 'void loop()':

Fire2012WithPalette:68: error: no matching function for call to 'random()'

   random16_add_entropy( random());

                                ^

C:\Users\Chris\Documents\Arduino\libraries\FastLED\examples\Fire2012WithPalette\Fire2012WithPalette.ino:68:32: note: candidates are:

In file included from sketch\Fire2012WithPalette.ino.cpp:1:0:

C:\Users\Chris\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:271:6: note: long int random(long int)

 long random(long);

      ^

C:\Users\Chris\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:271:6: note:   candidate expects 1 argument, 0 provided

C:\Users\Chris\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:272:6: note: long int random(long int, long int)

 long random(long, long);

      ^

C:\Users\Chris\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:272:6: note:   candidate expects 2 arguments, 0 provided

exit status 1
no matching function for call to 'random()'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
**************************************************************************************************************

what am i doing wrong here?

focalintent, 2017-01-04:
It sounds like the esp8266 library doesn't provide the standard random() function.  You can just comment out that line.

closed 2017-01-04 by focalintent

tablatronix, 2017-09-24:
esp8266 provides random , just not with empty arguments, and rand() cstdlib

https://github.com/esp8266/Arduino/issues/1283