FastLED/FastLED #391: APA106 is blinking with random colors

ollio, 2016-12-28:
Hey,

I'm experimenting with some LED's of type APA106. Specs to the LED's are [here](https://ae01.alicdn.com/kf/HTB12VXfLpXXXXc2XVXXq6xXFXXXV/222709402/HTB12VXfLpXXXXc2XVXXq6xXFXXXV.jpg?size=110147&height=801&width=708&hash=e2f86b4632b248be8a4a01d925f2a994). I already saw the issue [#298](https://github.com/FastLED/FastLED/issues/298) and tried to setup my LED's with the Alias to:
`
FastLED.addLeds<APA106, LED_PORT>(leds, NUM_LEDS);
`
The board I'm using is a NodeMCU (espressif8266) with 80MHz. Not sure if this is a timing issue.

If I run my test code (blink RED, GREEN, BLUE every second), than the LED's blinks in random colors.

[Video](https://goo.gl/photos/LoG6GgY9RFfqXjeR8)

ollio, 2016-12-29:
The problem was, that the wire between the two LED's was too long. Now I have it working with my own Controller definition: 

`
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class MyAPA106 : public ClocklessController<DATA_PIN, NS(350), NS(1000), NS(350), RGB_ORDER> {};
`

closed 2016-12-29 by ollio

platon-actual, 2019-02-12:
Hi.
To clarify some things:
FIRST: add the header FastLED:
`#include <FastLED.h>`
SECOND:
Add up the code below:
`template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class MyAPA106 : public ClocklessController<DATA_PIN, NS(350), NS(1000), NS(350), RGB_ORDER> {};`
This worked for me, trying an APA106 array of 64 NeoPixel LED's.
Bye.

sschueller, 2019-07-20:
FYI, this has been added to the Library and can be used by just setting LED_TYPE = APA106 in for example: https://github.com/FastLED/FastLED/blob/master/examples/ColorPalette/ColorPalette.ino