FastLED/FastLED #393: Arduino M0/ M0 Pro

tfach, 2017-01-09:
Hello, since the last topic on this issue was closed #292 (due to not answering) i'm trying it again. So: last time was asked which Arduino IDE was used: 
I'm working with Arduino IDE 1.8 on OSX. When Compiling i get the following Errors:

`In file included from /Users/XXX/Documents/Arduino/Cylon-APA102/Cylon-APA102.ino:1:0:
/Users/XXX/Documents/Arduino/libraries/FastLED/FastLED.h:558:2: warning: #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info." [-Wcpp]
 #warning "No pin/port mappings found, pin access will be slightly slower. See fastpin.h for info."
  ^
In file included from /Users/XXX/Documents/Arduino/Cylon-APA102/Cylon-APA102.ino:1:0:
/Users/XXX/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /Users/XXX/Documents/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /Users/XXX/Documents/Arduino/Cylon-APA102/Cylon-APA102.ino:1:
/Users/XXX/Documents/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /Users/XXX/Documents/Arduino/libraries/FastLED/FastLED.h:48:0,
                 from /Users/XXX/Documents/Arduino/Cylon-APA102/Cylon-APA102.ino:1:
/Users/XXX/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<12u>':
/Users/XXX/Documents/Arduino/libraries/FastLED/fastspi_bitbang.h:22:49:   required from 'class AVRSoftwareSPIOutput<12u, 11u, 4u>'
/Users/XXX/Documents/Arduino/libraries/FastLED/fastspi.h:30:7:   required from 'class SPIOutput<12u, 11u, 4u>'
/Users/XXX/Documents/Arduino/libraries/FastLED/chipsets.h:94:6:   required from 'class LPD8806Controller<12u, 11u, (EOrder)10u, 4u>'
/Users/XXX/Documents/Arduino/libraries/FastLED/FastLED.h:246:77:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)5u; unsigned char DATA_PIN = 12u; unsigned char CLOCK_PIN = 11u; EOrder RGB_ORDER = (EOrder)10u]'
/Users/XXX/Documents/Arduino/Cylon-APA102/Cylon-APA102.ino:27:71:   required from here
/Users/XXX/Documents/Arduino/libraries/FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
/Users/XXX/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<11u>':
/Users/XXX/Documents/Arduino/libraries/FastLED/fastspi_bitbang.h:23:50:   required from 'class AVRSoftwareSPIOutput<12u, 11u, 4u>'
/Users/XXX/Documents/Arduino/libraries/FastLED/fastspi.h:30:7:   required from 'class SPIOutput<12u, 11u, 4u>'
/Users/XXX/Documents/Arduino/libraries/FastLED/chipsets.h:94:6:   required from 'class LPD8806Controller<12u, 11u, (EOrder)10u, 4u>'
/Users/XXX/Documents/Arduino/libraries/FastLED/FastLED.h:246:77:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)5u; unsigned char DATA_PIN = 12u; unsigned char CLOCK_PIN = 11u; EOrder RGB_ORDER = (EOrder)10u]'
/Users/XXX/Documents/Arduino/Cylon-APA102/Cylon-APA102.ino:27:71:   required from here
/Users/XXX/Documents/Arduino/libraries/FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified
exit status 1
Error compiling for board Arduino M0 Pro (Programming Port).`

I have the board Arduino M0 Pro (Programming Port) selected.

focalintent, 2017-01-09:
Turn on verbose compiling output in the arduino settings and put the log in gist and link to it here - I want to see what arduino is defining for the platform - for whatever reason, what it's setting is not including the samd21 chip definitions.  (Note:  I'm still mostly on hiatus from this project due to a death in the family, but if there's something quick here I can see about getting it done - also, I have not done any work with the arduino IDE since 1.6.xx, because of said death).

tfach, 2017-01-09:
Hello, thank you for your fast response, here is the link to the verbose output (Arduino --> settings --> show verbose output during compilation): https://gist.github.com/tfach/3a52431b395a1d935d3dd88260be0c14
Sorry for your loss, this issue is not in a hurry.

tfach, 2017-01-11:
I updated the Fastled version and posted a new Debug output. This time with Windows and Arduino 1.8.1.

focalintent, 2017-01-11:
Ah - I see the problem - they're defining ARDUINO_SAM_ZERO -- however I only have pin definitions for ARDUINO_SAMD_ZERO, ARDUINO_SAMD_WINO, and ARDUINO_SAMD_MKR1000 (see platforms/arm/d21/fastpin_arm_d21.h -- for laughs, try adding the line:

```
#define ARDUINO_SAMD_ZERO
```

before `#include <FastLED.h>` 

not 100% sure that'll do what's expected -- but if the port/pin mappings are different on the arduino m0/m0pro than the arduino zero, then it won't work and i'll have to make up a new set of pin/port mappings at some point.

tfach, 2017-01-12:
Done compiling. Wow that was fast. It is working now. thank you 👍

closed 2017-01-12 by tfach

papperone, 2017-04-27:
Hello, just installed FastLED library and I've exactly the same issue and cannot compile any sketch netiher the example ones. My modules are ESP8266.
I use Arduino IDE 1.6.12 with latest board library ESP8266 2.2.3

jasoncoon, 2017-04-27:
@papperone, copy the error(s) you're getting, paste it in https://gist.github.com or https://pastebin.com and post link to it here.

papperone, 2017-04-27:
I updated Arduino IDE just in case to the latest 1.8.2 but error is still there
https://pastebin.com/fk1tmHcZ

jasoncoon, 2017-04-27:
Those aren't errors, they're pragma messages.  The compile completed successfully.

papperone, 2017-04-27:
thanks Jason, just for the sake of understading:
1) with previous IDE (1.6.12) the compile was not completed
2) you are right, compile is now completed but I still can't understand why I should get "red line" (which in standard language of colors are "ERRORS") if all is done correctly. 
can you explain a bit more on point 2) and what is the purpouse of those "red text" at the end?

jasoncoon, 2017-04-27:
1. Sorry, not sure.
2. See [any](https://github.com/FastLED/FastLED/issues/244) [of](https://github.com/FastLED/FastLED/issues/354) [these](https://github.com/FastLED/FastLED/issues/363) [conversations](https://plus.google.com/u/0/+AlexOzer/posts/1VM9vGbmyJV) for more info.  You can disable them by including #define FASTLED_INTERNAL before #include <FastLED.h>

papperone, 2017-04-27:
Thanks Jason :)

focalintent, 2017-04-27:
Also earlier today mark filed a bug with arruino folks to _not_ mark message output as an error - it is a bug on their part that they do.

kriegsman, 2017-04-27:
Here's the bug reported to the Arduino IDE folks: https://github.com/arduino/Arduino/issues/6227