FastLED/FastLED #394: Flickering Led Strip

pipoblak, 2017-01-09:
Hi,
i'm having some throubles with my LED setup, sometimes the LED STRIP get flicker.
I'm using a NODEMCU DEVKIT, 5v PowerSupply. Basiclly a NodeMcu and Strips  Connected  on a power Supply.

My Code : 
https://github.com/pipoblak/WebSocketAndFastLeds/blob/master/WebSocketWithFastLeds/WebSocketWithFastLeds.ino

Photos From My Setup: 
[image]
[image]
[image]

Daniel-dk, 2017-01-19:
What does the flicker look like ?
 * are all the ground signals connected ?
 * it could be due to signal issues on long wires ( do you have a 3v3v to 5V converter for pixel data  ? )
 * it could be dropped or out of sync frames ( if you stream frames to them over Wifi  - try to stick to UDP so teh retransmitted framse from TCP don't cause unwanted sparkly things)

editkid, 2017-01-21:
Is it this issue: https://github.com/FastLED/FastLED/issues/306?

If so, try `#define FASTLED_ALLOW_INTERRUPTS 0` before including FastLED. It may cause the NodeMCU to do watchdog resets though. You can read more about that in the linked issue.

pipoblak, 2017-01-22:
The flicker happens without explanation, after a undefined time the strip flicker a little and turns normal again.
All grounds is right conected.
I use a long wires between the signal and  NODEMCU , maybe can be that? 
And i dont use  3v3v to 5v converter for signal... .-.
The 3v3v converter can resolve my problem, maybe? 

Thx for all replys =D

editkid, 2017-01-22:
I couldn't only get 2-3 LEDs to light up without using a logic converter to up the control signal to 5V. I'm using a WeMos D1 Mini Pro though, not a NODEMCU.

Daniel-dk, 2017-01-23:
yes, long wires and low voltage can cause the signal to be degraded by noise ( or at least make teh effects of noise more apparent ).

you can use a WS2812B LED as a kind of signal step up, but a voltage converter to 5V is advised.
if you can, try to run your signal cable seperate from teh power cable ( AC power cables, high current DC etc ) can induce noise in your data cable.

closed 2017-05-29 by pipoblak