FastLED/FastLED #395: Auto Mode

LedArts, 2017-01-10:
Can anybody add 'auto mode' function. By which Automatically effects are change one by one with some delay.

kriegsman, 2017-01-11:
Check out the example called "DemoReel100" for one way to do this. This example is included in the library by default. You can use it as-is, or add more animations if you wish. 
Closing the ticket.

closed 2017-01-11 by kriegsman

LedArts, 2017-01-11:
thanks a lot sir, will try this.
Sir, there are six effects in demo Reel100 {rainbow, rainbowWithGlitter, confetti, sinelon, juggle, bpm}
can u tell me how much effects are present in fast led for pixal led strip.
i want run all the effects one by one in pixal led strip. please tell and give me all the effects of fast led with one program. I am using arduino with ic 328. Please sir, It will be more help full for me.

focalintent, 2017-01-11:
FastLED is not a library of LED effects.  It is a library for _making_ LED effects - as such, there's no such thing "all the effects" in FastLED - DemoReel100 attempts to show off some of the types of things you can do.   If you have specific coding questions, please ask on the g+ group at http://fastled.io/+

LedArts, 2017-01-12:
ok thank you sir...i have a one question. If i have to add more effects in demoReel100 how i add and which software should i use arduino or AVR Studio.

DeuxVis, 2017-01-12:
@LedArts in case that wasn't clear in @focalintent previous answer, you should **please ask on the g+ group at http://fastled.io/+** for all your questions.

Here - github - your will only get answers about "bug reports, feature requests and library changes".

Thanks !

alvin666666, 2017-04-20:
Hello,

We have one new led strip,gs8206,first it have the breakpoint continue ,then,the full color same with ws2811. Another important is,that have 33 kinds of color in the gs8206ic,no need the controller. but,if you connect the controller like ws2811,also is ok. It can be 12v and 24v. If you want to know more,send email to me.
Email:Alvin@witop-tech.com

kasperkamperman, 2017-04-20:
Is it possible to mark this Alvin guy as spam somewhere?

DeuxVis, 2017-04-20:
@kasperkamperman : Open its profile - clicking his icon or name - and use the "block or report user" feature there.