FastLED/FastLED #398: OctoWS2811Demo.ino does not work with 1024 LEDs per strip Teensy 3.2

il0vew33d, 2017-01-12:
hello everybody,

i'm going to build a huge ws2812 display with 128x64 leds.
yes, 8192 leds. for easier handling i use those 16x16 flexible ws2812 panels.
actually i have 16 panels put together to a 64x64 panel. i buy those 16x16 panels step by step, cause they are not so cheap. found them at aliexpress for about 35€ including shipping, thats ok for me.

for controlling my leds i use a teensy 3.2 with adafruit octows2811 breakout. successfully tested my 64x64 panel with the glediator sketch from octows2811 library and JINX!. the whole display works fine.

now i would program my own stuff and i tried out the fastled lib with octows2811 lib, used the sketch in examples/Multiple/OctoWS2811Demo with those parameters:

#define USE_OCTOWS2811
#include<OctoWS2811.h>
#include<FastLED.h>

#define NUM_LEDS_PER_STRIP 1024
#define NUM_STRIPS 8

CRGB leds[NUM_STRIPS * NUM_LEDS_PER_STRIP];

// Pin layouts on the teensy 3:
// OctoWS2811: 2,14,7,8,6,20,21,5

..
..
..

it compiles and i can upload the sketch to teensy, but the leds keep dark.
so i played a while with the value for NUM_LEDS_PER_STRIP and find out, the limit is at 800.

i guess there is a memory overflow or something like this.

focalintent, 2017-01-13:
FastLED double buffers the DMA output (so that you can call show and have the bits for the next frame get rotated out for OctoWS2811's format while the previous frame is still being written out).  So this means in total there are 3 full sets of memory for your leds.  There's the leds array that you defined above - which is 24k (1024 * 8 * 3 -- there are three bytes of rgb data per led) -- then there's two more buffers that size that are for OctoWS2811's data output - also 24k each - which is a total of 72k, but the Teensy 3.2 only has 64kb of ram, so you are running out of ram.  

I eventually hope to provide an option for single buffering the DMA block for OctoWS2811, to deal with memory situations like this - but it's unlikely to happen in the near future.

In the meantime, you don't have to use the OctoWS2811 driver for driving output, you can use the regular, non-DMA'd parallel output as described here - https://github.com/FastLED/FastLED/wiki/Parallel-Output

il0vew33d, 2017-01-13:
ok, i understand. thank you for the quick answer.
i will try the paralell output.

alvin666666, 2017-04-20:
why you use ws2812 display with 128x64 leds?  ws2813 is more reliable than ws2812,ws2813 is breakpoint continue. or we have sk9822,no matter what led broken,the whole led strip can work normally,factory price and high quality.
Email:Alvin@witop-tech.com

closed 2017-08-16 by focalintent