FastLED/FastLED #405: First color channel only causing serial issues.

Intrinsically-Sublime, 2017-01-22:
Platform: Arduino Pro Mini 328p running at 8mhz 
LEDs: 37 x WS2801
Compiled on Ubuntu 14.04 using Arduino 1.1.0.5
FastLED version 3.001.005

Issue: Whenever the Hue is set below 96 (0-95) or above 160 (161-255) the program continues running but it becomes completely unresponsive to Serial input.

I've tried to remove as much irrelevant code as possible. The problem seems to stem from two of the for loops in the chase() function I wrote. They set every 10th LED to be shown. If I remove these two for loops an only set one LED to be shown it works as expected. I have marked the two for loops with ******Possible cause of issue****** so you can easily find them.
 
```
#include "FastLED.h"

// For led chips like Neopixels, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806, define both DATA_PIN and CLOCK_PIN
#define DATA_PIN 3
#define CLOCK_PIN 9
#define SerialSpeed 115200 // I've tried speeds from 9600 all the way to 115200

// How many LEDs in your strip?
#define NUM_LEDS 37		// Total number of LEDs

// How are the LEDs distributed?
#define SegmentSize 5		// How many LEDs in each "Magnetic Constrictor" segment
#define TopLEDcount 20		// LEDs above the "Reaction Chamber"
#define ReactionLEDcount 2	// LEDs inside the "Reaction Chamber"
#define BottomLEDcount 15	// LEDs below the "Reaction Chamber"

// Default Settings
byte Warp_Factor = 1;	// 1-9
byte MainHue = 160;	// 1-255	1=Red 32=Orange 64=Yellow 96=Green 128=Aqua 160=Blue 192=Purple 224=Pink 255=Red
byte Saturation = 255;	// 1-255
byte Brightness = 96;	// 1-255

// Define internal parameters
#define TopSegments TopLEDcount/SegmentSize
#define BottomSegments BottomLEDcount/SegmentSize
#define PulseLength SegmentSize*2
#define TopChases TopLEDcount/PulseLength+1*PulseLength
#define BottomChases BottomLEDcount/PulseLength+1*PulseLength
#define TopLEDtotal TopLEDcount+ReactionLEDcount
#define BottomLEDtotal BottomLEDcount+ReactionLEDcount
#define TopDiff TopLEDcount-BottomLEDcount
#define RateMultiplier 6

// Initialize internal variables
byte ReactorHue = MainHue;
byte LastHue = MainHue;
byte Rate = RateMultiplier * Warp_Factor;	// RateMultiplier of 6 = range 6-54
byte Top;
byte Bottom;

// Serial input variables
const byte numChars = 20;
char receivedChars[numChars];
char tempChars[numChars];	// temporary array for use when parsing

// Parsing variables
byte warp_factor = Warp_Factor;
byte hue = MainHue;
byte saturation = Saturation;
byte brightness = Brightness;
boolean newData = false;

// Define the array of LEDarray
CRGB LEDarray[NUM_LEDS];

void setup() {
        delay(2000); // 2 second delay for recovery
	Serial.begin(SerialSpeed);
	FastLED.addLeds<WS2801,DATA_PIN, CLOCK_PIN, RGB>(LEDarray,NUM_LEDS);
//	FastLED.setCorrection(Typical8mmPixel);	// (255, 224, 140)
//	FastLED.setCorrection(TypicalSMD5050);	// (255, 176, 240)
	FastLED.setCorrection( CRGB( 255, 200, 245) );
	FastLED.setMaxPowerInVoltsAndMilliamps(5,1000);
	FastLED.setBrightness(Brightness);
	PrintInfo();
}

void loop() {
	receiveSerialData();
	if (newData == true) {
		strcpy(tempChars, receivedChars); // this is necessary because strtok() in parseData() replaces the commas with \0
		parseData();
		updateSettings();
		newData = false;
	}

	chase();
}

void chase() {
	for(int pulse = 0; pulse < PulseLength; pulse++) { // Default PulseLength = 10
		// Ramp LED brightness
		for(int value = 32; value < 255; value = value + Rate) {
			// Set every Nth LED in the top section   ******Possible cause of issue******
			for(int chase_top = 0; chase_top < TopChases; chase_top = chase_top + PulseLength) {
				Top =  pulse + chase_top;
				if (Top < TopLEDtotal) {
					LEDarray[Top] = CHSV(MainHue, Saturation, value);
				} else {
					break;
				}
			}
			// Set every Nth LED in the bottom section   ******Possible cause of issue******
			for(int chase_bottom = 0; chase_bottom < BottomChases; chase_bottom = chase_bottom + PulseLength) {
				Bottom = NUM_LEDS + TopDiff - (pulse + chase_bottom);
				if (Bottom > TopLEDcount) {
					LEDarray[Bottom] = CHSV(MainHue, Saturation, value);
				} else {
					break;
				}
			}
			// Keep reaction chamber at full brightness even though we chase the led right through it
			for (int reaction = 0; reaction < ReactionLEDcount; reaction++) {
				LEDarray[TopLEDcount + reaction] = CHSV(ReactorHue, Saturation, 255);
			}
			fadeToBlackBy( LEDarray, NUM_LEDS, (Rate*0.5));	// Dim all previous LEDs by Rate/2
			FastLED.show();					// Show set LEDs
		}
	}
}

void receiveSerialData() {
	static boolean recvInProgress = false;
	static byte ndx = 0;
	char startMarker = '<';
	char endMarker = '>';
	char helpMarker = '?';
	
	char rc;

	while (Serial.available() > 0 && newData == false) {
		rc = Serial.read();

		if (rc == helpMarker) {
			PrintInfo();
		} else if (recvInProgress == true) {
			if (rc != endMarker) {
				receivedChars[ndx] = rc;
				ndx++;
				if (ndx >= numChars) {
					ndx = numChars - 1;
				}
			}
			else {
				receivedChars[ndx] = '\0';
				recvInProgress = false;
				ndx = 0;
				newData = true;
			}
		} else if (rc == startMarker) {
			recvInProgress = true;
		}
	}
}

void parseData() {
	char * strtokIndx;			// this is used by strtok() as an index
	strtokIndx = strtok(tempChars,",");	// get the first part of the string
	warp_factor = atoi(strtokIndx);		// convert this part to an integer
	strtokIndx = strtok(NULL, ",");		// this continues where the previous call left off
	hue = atoi(strtokIndx);
	strtokIndx = strtok(NULL, ",");
	saturation = atoi(strtokIndx);
	strtokIndx = strtok(NULL, ",");
	brightness = atoi(strtokIndx);
}

void updateSettings() {
	if (warp_factor > 0 && warp_factor < 10 && warp_factor != Warp_Factor) {
		Warp_Factor = warp_factor;
		Serial.print("Warp Factor Set To ");
		Serial.println(warp_factor);
	}
	if (hue > 0 && hue < 256 && hue != MainHue) {
		MainHue = hue;
		ReactorHue = hue;
		LastHue = hue;
		Serial.print("Color Hue Set To ");
		Serial.println(hue);
	}
	if (saturation > 0 && saturation < 256 && saturation != Saturation) {
		Saturation = saturation;
		Serial.print("Color Saturation Set To ");
		Serial.println(saturation);
	}
	if (brightness > 0 && brightness < 256) {
		FastLED.setBrightness(brightness);
		Brightness = brightness;
		Serial.print("Brightness Set To ");
		Serial.println(brightness);
	}
	Rate = RateMultiplier * Warp_Factor;
}

void PrintInfo() {
	Serial.println("");
	Serial.println("******** Help ********");
	Serial.println("Input Format - <2, 160, 220, 255>");
	Serial.println("Input Fields - <Warp Factor, Hue, Saturation, Brightness>");
	Serial.println("Warp Factor range - 1-9");
	Serial.println("Hue range - 1-255 1=Red 32=Orange 64=Yellow 96=Green 128=Aqua 160=Blue 192=Purple 224=Pink 255=Red");
	Serial.println("Saturation range - 1-255");
	Serial.println("Brightness range - 1-255");
	Serial.println("**********************");
	Serial.println("");
	Serial.println("** Current Settings **");
	Serial.print(" <");
	Serial.print(Warp_Factor);
	Serial.print(", ");
	Serial.print(MainHue);
	Serial.print(", ");
	Serial.print(Saturation);
	Serial.print(", ");
	Serial.print(Brightness);
	Serial.println(">");
	Serial.println("**********************");
	Serial.println("");
}
```

kriegsman, 2017-01-22:
Hi and thanks for using FastLED. For general "what's going wrong here in this code?" questions, we discuss them over on the Google Plus community for FastLED: http://FastLED.io/+
We like to keep the github issue tracker for bugs in the library itself, and I think this one is more of a general memory corruption bug or similar. 
Please do post your question over on G+, and if it turns out to be a library bug, we can reopen this ticket!
(Closing the github ticket for now though)

closed 2017-01-22 by kriegsman

Intrinsically-Sublime, 2017-01-22:
Thanks for taking the time to respond rather than just close or ignore it but like many of us in the world there is no chance I will ever be on G+

kriegsman, 2017-01-22:
Yep. I hear you.  Hope you get the project debugged, and the community is there if you even just want to browse it. Thanks for writing.