FastLED/FastLED #406: ATSAMD21E17A Invalid pin specified even with Software SPI

Knoskokoloz, 2017-01-27:
Hello + thanks for creating fastled really added some nice effect for some of my cosplay props :)

 I have some troubles with FastLED version 3.001.005/Arduino IDE 1.8.1 and an Atmel ATSAMD21E17A (pretty much the CPU of the Arduino Zero with half Ram and Flash). 
http://rabidprototypes.com/product/firecricket/ this is the board I use.

For testing I used the example ColorPalette. And ended up with the compile error message 
"static_assert(validpin(), "Invalid pin specified");"
I thought maybe even if the cpu s similiar to the zero it might need a specific Pin Layout for hardware spi 
and tried.
#define FASTLED_FORCE_SOFTWARE_SPI
as described in https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging
and still ended with "static_assert(validpin(), "Invalid pin specified");"

Do I still need an extra pin layout for Software SPI or am I missing something?

I also tried to make a pinlayout at fastpin_arm_d21.h but sadly no Idea what to do.

Thanks in advance if someone can help me with this

Knoskokoloz, 2017-02-20:
Got it working just replaced #if defined(ARDUINO_SAMD_ZERO) with #if defined(_SAMD21E17A_) in the file fastpin_arm_d21.h and seems to work just fine.

focalintent, 2017-02-20:
The reason for the check on samd_zero instead of the chip id is because the port to pin mapping isn't based on the chip, but on the board (though it's not uncommon for clone boards to use the same port to pin mapping, but I've been burned in the past by that)

Knoskokoloz, 2017-02-21:
Ah I see that makes sence. Actually I think the port mapping is a bit different as seen here
http://rabidprototypes.com/wp-content/uploads/2016/09/Firecricket-Pinout.png
http://rabidprototypes.com/wp-content/uploads/2017/02/Firecricket-Schematic.png
At least for the number of available Pins. Sadly I'm a bit lost how to do a propper pin layout
_DEFPIN_ARM( 1,0,11);  <- I assume first number is the number of the pin but the other 2? 
SPI clock and data might be the same and thats why it simply works.

Thanks for the reply!

closed 2017-03-06 by Knoskokoloz