FastLED/FastLED #407: Changing argument for EVERY_N_MILLISECONDS() does not seem to work.

atuline, 2017-02-02:
I have a program with several demos, each requiring a different delay, so I've been using something like this:

EVERY_N_MILLISECONDS(thisdelay) { run_nextdemo(); }

Although I can change the value of 'thisdelay', the EVERY_N_MILLISECONDS will ONLY use very first value it saw. If I define a global variable:

int thisdelay = 0; 

at the start of the program, then changing it to another value has no effect when I call EVERY_N_MILLISECONDS(thisdelay) in my loop. It still assumes '0'.

I don't know if this is a problem with my program (maybe there's a memory overflow issue), a problem with EVERY_N_MILLISECONDS or just the way it's supposed to work. I'll make a short test program and use a beatsin8().

Blink515, 2017-02-02:
Andrew,
I had the same question a while back and was directed to this info from the
G+ community.  The EVERY_N_MILLISECONDS_I function works really well.

https://plus.google.com/106461657512143768817/posts/AyRzBp77g4h?sfc=false

atuline, 2017-02-02:
Thanks Blink. I've change it to:

  EVERY_N_MILLIS_I(thistimer, thisdelay) {
    strobe_lights(current_mode);
  }

  thistimer.setPeriod(thisdelay);

and it works like a charm. I'll need to throw that into my documentation for noobs (like me).

closed 2017-02-02 by atuline

marmilicious, 2017-02-02:
I'm surprised you didn't ask this in the G+ group Andrew.  :)  Here's an example I made some time back.
https://github.com/marmilicious/FastLED_examples/blob/master/every_n_timer_variable.ino

theiothing, 2018-03-20:
@atuline. 1st thing: if you are noob we are all more than ignorants.... 2nd ly. can you use setPeriod outside the function? I recieve timing inputs via MQTT, so I assign them to a global variable and, the only solution I have found  so far, was to handwrite the timing condition, like:
```
uint64_t c_millis = millis();
if ((uint64_t)(current_millis - previous_millis) >= timerCounter){
//do something
previous_millis = current_millis;
}
```
cheers

tonton81, 2018-03-20:
millis() is uint32_t, NOT uint64_t.

drf5n, 2023-02-16:
I used a hand-rolled millis()-last test to make a slide-potentiometer based interval to give adjustable speed in this Wokwi simulation

 https://wokwi.com/projects/356666059490147329