FastLED/FastLED #410: FirstLight Won't Compile with ESP8266 Huzzah

gfvalvo, 2017-02-10:
Hi. I'm really sorry to post this because I've seen lots of ESP8266 threads, but I'm lost reading my way through all of them. I'm trying to compile FirstLight on an Adafruit Huzzah (link below) connected to a strip of 120 Adafruit DotStars (two concatenated strips, link below, APA102 controller I believe). The strip is connected to the Huzzah's Hardware SPI pins and works fine with Adafuit's DotStar library. I want to use the same Hardware SPI with fastLed.

So, first question, does fastLed now support Hardware SPI on this board? (I downloaded the fastLed library on 9 February, 2107).

If so, here's my much stripped down sketch that won't compile:
```
#include "FastLED.h"

#define NUM_LEDS 120

CRGB leds[NUM_LEDS];

void setup() {
  delay(2000);
  FastLED.addLeds<DOTSTAR, RGB>(leds, NUM_LEDS);
}

void loop() {
  for (int whiteLed = 0; whiteLed < NUM_LEDS; whiteLed = whiteLed + 1) {
    // Turn our current led on to white, then show the leds
    leds[whiteLed] = CRGB::White;

    // Show the leds (only one of which is set to white, from above)
    FastLED.show();

    // Wait a little bit
    delay(100);

    // Turn our current led back to black for the next loop around
    leds[whiteLed] = CRGB::Black;
  }
}
```
And here are the error messages:
```
Arduino: 1.6.12 (Windows 7), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_963529\FirstLight.ino:1:0:

D:\GFV's Documents\Computer Stuff\Arduino Projects\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.005

 #    pragma message "FastLED version 3.001.005"

                     ^

In file included from D:\GFV's Documents\Computer Stuff\Arduino Projects\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_963529\FirstLight.ino:1:

D:\GFV's Documents\Computer Stuff\Arduino Projects\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_963529\FirstLight.ino: In function 'void setup()':

FirstLight:9: error: no matching function for call to 'CFastLED::addLeds(CRGB [120], int)'

   FastLED.addLeds<DOTSTAR, RGB>(leds, NUM_LEDS);

                                               ^

C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_963529\FirstLight.ino:9:47: note: candidates are:

In file included from C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_963529\FirstLight.ino:1:0:

D:\GFV's Documents\Computer Stuff\Arduino Projects\libraries\FastLED/FastLED.h:218:129: note: template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN, EOrder RGB_ORDER, unsigned char SPI_DATA_RATE> CLEDController& CFastLED::addLeds(CRGB*, int, int)

  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN, EOrder RGB_ORDER, uint8_t SPI_DATA_RATE > CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
[... 111 lines cut ...]
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {

                         ^

D:\GFV's Documents\Computer Stuff\Arduino Projects\libraries\FastLED/FastLED.h:445:25: note:   template argument deduction/substitution failed:

FirstLight:9: error: could not convert template argument 'DOTSTAR' to 'EBlockChipsets'

   FastLED.addLeds<DOTSTAR, RGB>(leds, NUM_LEDS);

                                               ^

exit status 1
no matching function for call to 'CFastLED::addLeds(CRGB [120], int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```
Any help would be appreciated.
Thanks.
Links:
[https://www.adafruit.com/products/2821](url)
[https://www.adafruit.com/products/2239](url)

jasoncoon, 2017-02-10:
I don't think FastLED supports hardware SPI on ESP8266 yet.  Try adding the pin numbers to your call to FastLED.addLEDs.

gfvalvo, 2017-02-10:
Yea, I saw a post to that effect during my Google search. But, it was well over a year old. I was hoping that ESP8266 Hardware SPI support would have been added by now.

Anyway, thanks, it does work with Bit -Bang SPI. A shame though as I must now choose between the advanced features of fastLED and the Hardware SPI support of Adafruit's library.

focalintent, 2017-08-16:
Tracking hardware SPI for the ESP8266 in #429

closed 2017-08-16 by focalintent