FastLED/FastLED #411: Maximum WS2811 can be driven using Arduino uno Atmega 328

Saravananece82, 2017-02-14:
Iam planning to use WS2811 with Arduino uno Atmega 328 is it possible to use  for 5000 leds
Iam new to arduino, Let me know the driver code area for led

focalintent, 2017-02-14:
Not with a 328, with fastled.  The 328 only had 2kb of ram and each led requires a minimum of 3 bytes of ram - which would be 15k at least. 

Ws2811 leds take 30us to write out - 5000 will take 150ms which means you could only do 7fps top.  I'd strongly recommend you look at something like a teensy 3.2 and using parallel output.  Also, in the future please use the g+ (http://fastled.io/+) group for "how do I?" questions - I want to leave GitHub issues for bug reports.

closed 2017-02-14 by focalintent

Saravananece82, 2017-02-14:
Thanks for your quick reply