FastLED/FastLED #412: Blue dot in spectrum?

kriegsman, 2017-02-21:
From G+
https://plus.google.com/u/0/102302399546211644090/posts/UoWjJcHrp6B?cfem=1

Hi!

I am using FastLED in combination with the hsv2rgb_spectrum() function. While cycling a rainbow pattern, I noticed a blue dot running across the LEDs.

https://www.youtube.com/watch?v=IYAWEHtRvAE

After further investigation it turned out that when CHSV(255, 255, 255) (red) is passed to hsv2rgb_spectrum(), consistently results in 0, 0, 251 (blue). Regardless of the saturation or value, if the hue is 255, red remains 0 and blue changes accordingly. With a hue of either 254 or 256, the result will again be red, going round as expected.

    CHSV hsv = CHSV(255, 255, 255);
    CRGB rgb;
    
    hsv2rgb_spectrum(hsv, rgb); // 0, 0, 251

Am I misunderstanding how to use this function? Since CHSV(255, 255, 255) is indeed red when used on its own, I would expect a result that approaches that when converted to spectrum RGB. Thank you for your reply!﻿

focalintent, 2017-02-21:
Another victim of fixing scale8?

kriegsman, 2017-02-21:
That's what I'm wondering, right?
Some times I think maybe I should save all those unit-test-y things that I've written over the years, and, you know, run them or something.
Anyway, I'll take a look at this one.

commit 77416b5d (2017-02-22): Updated hsv2rgb_spectrum for the fixed scale8.  Fixes #412
closed 2017-02-22 by kriegsman