FastLED/FastLED #413: Making FastLED 3.1.3 work with ATTinyCore

telanoc, 2017-02-24:
I tried a simple project built with a bare ATtiny85 and a neopixel strip in Arduino 1.8.1 with the ATTinyCore board support 1.1.2 added and got the problem of yield() and timer0_millis not being found, much like other people have seen. Being stubborn, I dug in and fixed the two to my satisfaction, but since the only ATtiny chip I have is the '85, I can't really tell if this is the good and proper fix.

I have a gist at https://gist.github.com/telanoc/ec28d6b641a01c9bf2afd0784dd9f28c with the two changes in 'diff -u' form, if anyone is interested.  Basically: The ATTinyCore uses 'millis_timer_millis' as its counter variable, and don't call yield when ATTINY_CORE is defined.

jipp, 2017-05-19:
Hi

I have the same problem using platfromIO and compiling fastLED against an ATTiny85:

.piolibdeps/FastLED_ID126/FastLED.cpp: In member function 'void CFastLED::delay(l

ong unsigned int)':
.piolibdeps/FastLED_ID126/FastLED.cpp:133:9: error: 'yield' was not declared in t
his scope
yield();
^
*** [.pioenvs/attiny85-16MHz/lib/FastLED_ID126/FastLED.o] Error 1

GizmoTheGreen, 2018-07-19:
the changes to led_sysdefs_avr.h by telanoc in combination with FASTLED_NEEDS_YIELD got FastLED lib work for me on attiny 85! :)

why is the timer_millis fix not in FastLED yet?

focalintent, 2018-07-19:
See this on the main github page of  the project:

> *NOTE* library work is currently on hold - see https://plus.google.com/102282558639672545743/posts/JQKcujjaXXC for details. 

and there’s no pull request for this - i’ve been merging in pull requests as I see them and they seem safe enough, but I don’t really have an offline environment at the moment where I can juggle/merge code by hand outside of the github UI.

telanoc, 2018-07-19:
I forked the project to make a proper pull request but when I went to make the changes in my local copy, it looks like things are already taken care of.  I had made two changes in my gist, but the ATTINY_CORE ifdef is already in your code, and the yield() call isn't flagged as undefined.  Perhaps there were updates to the tiny core that fixed that bit.

To further test, I pulled a new attiny85 out of the tube, put it in the programmer with a strip of LEDs attached and threw a strand-test project at it.  It's working fine in Arduino with FastLED 3.1.6.  I guess I should close this?

closed 2018-07-19 by telanoc

GizmoTheGreen, 2018-07-20:
that's really weird, I just installed both attinycore and fastled(3.1.6) yesterday and it wouldn't compile until I added your first fix and the define to activate the yield code... :/

not sure if it's a bug on arduino side or the uploaded version thats listed in library manager but the fixes were not in it.

Puneethkumarsn, 2020-12-05:
Really Sorry if I am spamming

Here is a  link from wokwi Arduino simulator

https://wokwi.com/arduino/projects/283910810787381773?ck_subscriber_id=972903999