FastLED/FastLED #414: FastLED doesnt work with Arduino MKRZero

javierru3, 2017-02-27:
I tried to compile a working program with FastLED and Arduino UNO/Pro Mini on MKRZero and it doesn't work. I become the following warning:
#warning "No pin/port mappings found, pin access will be slightly slower.
and a lot of compilation errors that didn'nt appear when i choose UNO in Boards. It's Known when Arduino MKRZero will be suported? Could i help in the process? or it's easier to rewrite the program without Library?

Daniel-dk, 2017-02-27:
the MKRZero has a SAMD21 ( ARM  ) on it, not a Atmega 328 ( Uno / Pro mini )
if the  pin/port mapping is differnet from the arduino Zero You'd have to add the pin and port difinitions to the files in FastLED\platforms\arm\d21 

if they are the same then you can simply use the Zero as your board selection 
( dont have time to add it at the moment, might get some near the weekend to check pin/port mappings for Zero vs MKRZero )

javierru3, 2017-02-28:
Thanks I will try to do it! If the pinports in Zero and MKRZero are the same, I shouldn't have problems if I compile the program with zero configuration... Am I right?

javierru3, 2017-05-03:
Hello,
I have been trying for a while to compile my Skript using  Zero as my board selection, and it works, with the only problem that the pins on Arduino IDE arent the same as in the board.
I would like to use the SD slot on my MKRZero, but since i am not using the MKRZero as a board selection i can not use the MKRZero command "SD.begin(SDCARD_SS_PIN)" there is a way to know which CS should i use to select my sd card?

Daniel: I tryed to add the pin definitions to FastLED but i havent the needed knowledge :/
Thanks

WyattJordan, 2017-11-25:
javierru3

Did you ever figure out how exactly the FastLED\platforms\arm\d21\fastpin_arm_d21 file needed to be changed? I just ran into the exact same problem as you and have done a TON of work on a project that requires the MKRZero to use the FastLED library (with the SD card) and would greatly appreciate a solution if you found one, thanks!

WyattJordan, 2017-11-26:
For anyone who runs into this problem in the future I found a solution. Go to the FastLED\platforms\arm\d21\fastpin_arm_d21.h file and add this code as one of the platform pin mapping cases:

`#elif defined(ARDUINO_SAMD_MKRZERO)
#define MAX_PIN 22
_DEFPIN_ARM(  0, 0, 22); _DEFPIN_ARM(  1, 0, 23); _DEFPIN_ARM(  2, 0, 10); _DEFPIN_ARM(  3, 0, 11);
_DEFPIN_ARM(  4, 1, 10); _DEFPIN_ARM(  5, 1, 11); _DEFPIN_ARM(  6, 0, 20); _DEFPIN_ARM(  7, 0, 21);
_DEFPIN_ARM(  8, 0, 16); _DEFPIN_ARM(  9, 0, 17); _DEFPIN_ARM( 10, 0, 19); _DEFPIN_ARM( 11, 0,  8);
_DEFPIN_ARM( 12, 0,  9); _DEFPIN_ARM( 13, 1, 23); _DEFPIN_ARM( 14, 1, 22); _DEFPIN_ARM( 15, 0,  2);
_DEFPIN_ARM( 16, 1,  2); _DEFPIN_ARM( 17, 1,  3); _DEFPIN_ARM( 18, 0,  4); _DEFPIN_ARM( 19, 0,  5);
_DEFPIN_ARM( 20, 0,  6); _DEFPIN_ARM( 21, 0,  7);
#define SPI_DATA 8
#define SPI_CLOCK 9

#define HAS_HARDWARE_PIN_SUPPORT 1`

I went through and did all the mapping and turns out its the same as the `#elif defined(ARDUINO_SAMD_MKR1000)` case, you just need to make a case for `#elif defined(ARDUINO_SAMD_MKRZERO)`

ubidefeo, 2017-12-04:
@WyattJordan thanks for this, I was wondering why it worked on the MRK1000 and not the ZERO (we also ran into some pin mapping issue with the Feather M0, same SAMD21).
It would be nice if you could issue a Pull Request for this, unless the developers already implemented the definition (in the PlatformIO version it's not working yet)

rogertheriault, 2018-02-13:
Thanks. Only one line needs to change, since the settings are the same
```
 #elif defined(ARDUINO_SAMD_MKR1000) || defined(ARDUINO_SAMD_MKRZERO)
```

adidahiya, 2018-12-06:
Thanks @WyattJordan and @rogertheriault.

For the MKR WIFI 1010, the code change is:

```c
#elif defined(ARDUINO_SAMD_MKR1000) || defined(ARDUINO_SAMD_MKRWIFI1010)
```

tigoe, 2019-01-11:
It seems to work well with:

````
#elif defined(ARDUINO_SAMD_VARIANT_COMPLIANCE)
````

as well. All the MKR boards in the SAMD core of Arduino feature that particular define, so it should work for MKR Zero, MKR 1000, MKR 1010, MKR 1400, etc.  I've only tested with the Zero, the 1000, and the 1010, though.

ubidefeo, 2019-01-11:
thank you @tigoe for pointing this out.
I've always manually defined `ARDUINO_SAMD_MKR1000` when using FastLED in the past.
It's nice that such define for generic SAMD exists :)

tigoe, 2019-01-11:
Don't quote me on it though, Check with @cmaglie to be sure.

ubidefeo, 2019-01-11:
@tigoe @cmaglie I went to check the builder's verbose messages and found that `ARDUINO_ARCH_SAMD` is defined during build (`-D ARDUINO_ARCH_SAMD`) hence this should be generic enough, but I haven't found trace of the definition you mention.
I've tested in Create production but not in the Java IDE

tigoe, 2019-01-11:
Yours is probably more correct, then. I was using the Java IDE, FWIW, and
found it in the variant.h file of the samd core.

t.

matthijskooijman, 2019-01-11:
But wouldn't using such a generic define mean you will apply the same pinout to *all* SAMD boards? Maybe all current SAMD boards use the same pinout, but this might not be true in the future (and it would be better for the compiler to throw an error on new boards that might or might not be supported, than to have no error and have Fastled toggle the wrong pins).

If one would want to catch all boards with the same pinout, than a define that identifies the variant used could be appropriate (since typically all boards using the same variant use the same pinout), but I'm not sure if the variant is available in a define? Also, there is nothing that would stop variant files from subtly modifiying the pinout based on other defines, so only checking the define might not be sufficient either (in the future).

ubidefeo, 2019-01-11:
@matthijskooijman all MKR boards have the same pinout and quite a high level of customisation to add/re-route SERCOMs and other hardware.
Different boards which have extra modules (Vidor, GSM, LoRa and so on) are connected to extra hardware using the other pins of the chip not routed to the main headers :)

tigoe, 2019-01-11:
For example, the SD on the MKRZero is routed to a different SPI SERCOM, if I recall correctly, which is why it works with the generic define, I assume.

@matthijskooijman's approach is a valid, conservative approach, though. It takes more maintenance, but it means any exceptions that pop up in the MKR line in the future are more likely to get addressed.

elektrowolle, 2019-04-03:
Nobody is talking about this anymore, but I made a fork a year ago, changing it exactly as @adidahiya did 00f17a84dab9105e40d0632ae614e90b034c75fb. Worked fine for me on many devices so far.

tigoe, 2019-09-25:
Just ran into a student from someone else's class that was trying to use this library with one of the MKR variants, and found it didn't work still. We had to modify the lib using #elif defined(ARDUINO_SAMD_VARIANT_COMPLIANCE) again. So it seems that these changes haven't yet made it to the current version.

PMHK, 2021-03-26:
Having updated to the latest FastLED library this morning, I have again had to modify the file at src/platforms/arm/d21/fastpin_arm_d21.h in order to be able to use the library with the Arduino MKR Zero board, I've done this by adding the Zero board in as an OR with the MKR1000 and MKRWIFI1010.

As this has been a known issue for a while, I'm wondering if there is a reason I haven't spotted that an update such as this or those described above haven't made it into the release, I'm slightly concerned that whilst this change seems to work for me there might be a reason why this isn't such a good idea that I'm not aware of?

sonatique, 2022-07-07:
Same as last comment from @PMHK, but now July 2022. Thanks in advance for replying

huster-songtao, 2023-11-10:
> Having updated to the latest FastLED library this morning, I have again had to modify the file at src/platforms/arm/d21/fastpin_arm_d21.h in order to be able to use the library with the Arduino MKR [...]

**How  to modify the file at src/platforms/arm/d21/fastpin_arm_d21.h in order to be able to use the library with the [Seeed Studio XIAO ESP32C3](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/)？**
[image]

ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038855e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x458
load:0x403cc710,len:0x814
load:0x403ce710,len:0x2878
entry 0x403cc710

The symbolic names for the 4 analogue pins
A0 = 2
A1 = 3
A2 = 4
A3 = 5

The symbolic names for the 11 digital pins
D0 = 2
D1 = 3
D2 = 4
D3 = 5
D4 = 6
D5 = 7
D6 = 21
D7 = 20
D8 = 8
D9 = 9
D10 = 10

**I can not use FastLED with  XIAO ESP32C3 board and WS2812B Led Strip.**

\Arduino\Sketchbook\libraries\FastLED\release_notes.md

> FastLED 3.4.0
> * Merged in contributed support for Seeedstudio Odyssey and XIAO boards

Arduino\Sketchbook\libraries\FastLED\src\platforms\arm\d21\fastpin_arm_d21.h

> #elif defined(SEEED_XIAO_M0)
> 
> #define MAX_PIN 10
> _FL_DEFPIN( 0, 2,0); _FL_DEFPIN( 1, 4,0); _FL_DEFPIN( 2,10,0); _FL_DEFPIN( 3,11,0);
> _FL_DEFPIN( 4, 8,0); _FL_DEFPIN( 5, 9,0); _FL_DEFPIN( 6, 8,1); _FL_DEFPIN( 7, 9,1);
> _FL_DEFPIN( 8, 7,0); _FL_DEFPIN( 9, 5,0); _FL_DEFPIN(10, 6,0);
> 
> #define SPI_DATA 9
> #define SPI_CLOCK 8
> 
> #define HAS_HARDWARE_PIN_SUPPORT 1
> 

**The following code, I used Adafruit_ NeoPixel works well with the WS2812B led strip, using the D10 PIN on the Arduino is connected to the NeoPixels**

```
// NeoPixel Ring simple sketch (c) 2013 Shae Erisson
// Released under the GPLv3 license to match the rest of the
// Adafruit NeoPixel library

#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
 #include <avr/power.h> // Required for 16 MHz Adafruit Trinket
#endif

// Which pin on the Arduino is connected to the NeoPixels?
#define PIN        10 // On Trinket or Gemma, suggest changing this to 1

// How many NeoPixels are attached to the Arduino?
#define NUMPIXELS 8 // Popular NeoPixel ring size

// When setting up the NeoPixel library, we tell it how many pixels,
// and which pin to use to send signals. Note that for older NeoPixel
// strips you might need to change the third parameter -- see the
// strandtest example for more information on possible values.
Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

#define DELAYVAL 100 // Time (in milliseconds) to pause between pixels

void setup() {
  // These lines are specifically to support the Adafruit Trinket 5V 16 MHz.
  // Any other board, you can remove this part (but no harm leaving it):
#if defined(__AVR_ATtiny85__) && (F_CPU == 16000000)
  clock_prescale_set(clock_div_1);
#endif
  // END of Trinket-specific code.

  pixels.begin(); // INITIALIZE NeoPixel strip object (REQUIRED)
}

void loop() {
  pixels.clear(); // Set all pixel colors to 'off'

  // The first NeoPixel in a strand is #0, second is 1, all the way up
  // to the count of pixels minus one.
  for(int i=0; i<NUMPIXELS; i++) { // For each pixel...

    // pixels.Color() takes RGB values, from 0,0,0 up to 255,255,255
    // Here we're using a moderately bright green color:
    pixels.setPixelColor(i, pixels.Color(0, 233, 255));
    pixels.show();   // Send the updated pixel colors to the hardware.

    delay(DELAYVAL); // Pause before next pass through loop
  }

  for(int i=0; i<NUMPIXELS; i++) { // For each pixel...

    // pixels.Color() takes RGB values, from 0,0,0 up to 255,255,255
    // Here we're using a moderately bright green color:
    pixels.setPixelColor(i, pixels.Color(0, 0, 0));
    pixels.show();   // Send the updated pixel colors to the hardware.

    delay(DELAYVAL); // Pause before next pass through loop
  }

}

```