FastLED/FastLED #415: FastLED.h and IRremote.h together - issue!

sg1oneill, 2017-02-28:
If I use FastLED.show(); anywhere in the code, it screws up the incoming code of the remote!
It took me a while to figure out what cause the problem, but I think FastLED.show(); function does.
However, I have no clue whatever it is a FastLED library problem or IRremote but anyway I share it here, maybe someone can fix it or help with it. Thanks
[Uploading IRemote_practice.ino.zip…]()

focalintent, 2017-02-28:
See https://github.com/FastLED/FastLED/wiki/Interrupt-problems for a discussion of the issue and things you can try to work around it.

closed 2017-02-28 by focalintent

sg1oneill, 2017-02-28:
Thank you but at first I cannot see anything in that thread that would help on this problem...
IRremote  does not work fine if I use (and I have to use) FastLED.show();  Is it IRremote or FastLED I don't know... 
Please, could you be a more specific as I cannot figure out from the thread what next...
Thank you.

kriegsman, 2017-03-01:
IRremote requires that Interrupts be turned ON _all of the time_ in order to receive signals properly.

WS2811 "Neopixel" LEDs require that Interrupts be turned **OFF** while sending data to the LEDs.

It's a fundamental conflict between the hardware: the IR receiver, and the LEDs.  You can't run them both reliably from a single Arduino microcontroller board.

For more help with this topic, try posting a question on our (much larger, much more active) Google Plus group for FastLED:  http://fastled.io/+

marcmerlin, 2017-04-17:
I'm seeding google results (including this one) with a solution page I wrote for this after a few days of work.
This page and video explain the issues between IR and neopixels, and how to make it work on teensy v3.1, ESP8266, and ESP32:
http://marc.merlins.org/perso/arduino/post_2017-04-03_Arduino-328P-Uno-Teensy3_1-ESP8266-ESP32-IR-and-Neopixels.html
this short video shows how trying this on AVR is an exercise in frustration and handling failures :)
https://www.youtube.com/watch?v=62-nEJtm070