FastLED/FastLED #417: Error addressing 7 strips of LEDs - WEMOS D1R2

StevenPorter, 2017-03-02:
I am working on a scrolling message board project using [7 strips of WS2812B leds](https://youtu.be/0mOBwzo0jpw). The code compiles on a UNO board, but fails to compile on the WeMos D1 R2 board.  I have tried both the stable and staging versions of the ESP8266 libs.  

When I compile with all 7 strips initialized I get the following error:

`C:\Users\sporter...\FastLED-WeMosD1R2.ino.elf section .text will not fit in region iram1 0 seg`

I am no expert with Arduino, but I don't think this is a memory issue.  It doesn't matter if I have 100 LEDs per strip or just 2 LEDs per strip.  It compiles successfully with 6 strips initialized and has plenty of memory to spare:

```
Sketch uses 228733 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32980 bytes (40%) of dynamic memory, leaving 48940 bytes for local variables. Maximum is 81920 bytes.
```

I have uploaded example code that complies as is with only 6 LEDs initialized, If you uncomment the last FastLED.addLeds.. statement it will fail the compile on the WeMos D1R2 board.

I would like to use the WeMos board to use the WiFi to remotely control the scrolling message.  Once I get past this issue I will post the entire codeset.  Any help from the team and/or community would be greatly appreciated!!

[FastLED-WeMosD1R2.txt](https://github.com/FastLED/FastLED/files/815116/FastLED-WeMosD1R2.txt)

focalintent, 2017-03-02:
The ws2811 code is flagged for being in, if I remember right, iram for better timing - unfortunately, having 7 or 8 copies of that maxes out the section, resulting in the error you're seeing.  Fixing that is on the list of things to do when I get back to working on the library, but there's really no time frame for that right now.

StevenPorter, 2017-03-03:
Appreciate the quick response!  Thank you!

Is there another LED model that would work better for running multiple strands? I know the fastLED library supports many, but I am not sure which one is best.

Thanks in advance!

-Steve

focalintent, 2017-03-04:
Not for running more than 6 or 7 pins, at least until I get to rewrite/change up that code a bit.  There's some experimental parallel output support but I don't remember what state I left it in and that only covers 4 pins (because of CPU overhead and pin/port layout, it wasn't possible to do 8 way parallel output like the teensy 3.x has).

StevenPorter, 2017-03-04:
OK.  Thanks again for the help.  Love the Library and all its functionality.  I will probably work toward wiring them up in series and doing the math in code to still write out what looks like 7 lines.  Should be a fun exercise.  Appreciate the help.

-Steve

coryking, 2017-12-16:
@focalintent , is there a workaround for this?

I'm getting this error with just 4 calls to addLeds on my ESP8266 (d1_mini, 4mb flash)

```c    
FastLED.addLeds<LED_TYPE, DATA_0_PIN, GRB>(player.getFastLEDBuffer(), 0, LEDS_PER_STRIP);
FastLED.addLeds<LED_TYPE, DATA_1_PIN, GRB>(player.getFastLEDBuffer(), LEDS_PER_STRIP, LEDS_PER_STRIP);
FastLED.addLeds<LED_TYPE, DATA_2_PIN, GRB>(player.getFastLEDBuffer(), 2*LEDS_PER_STRIP, LEDS_PER_STRIP);
FastLED.addLeds<LED_TYPE, DATA_3_PIN, GRB>(player.getFastLEDBuffer(), 3*LEDS_PER_STRIP, LEDS_PER_STRIP);
```

3 calls, this compiles fine.  4 and no dice.

Cheers!

focalintent, 2017-12-18:
No - I haven't started working on the library again, and fixing this requires some major rewriting.

skrenes, 2018-03-02:
I'm also having this issue (```section `.text' will not fit in region `iram1_0_seg'```) when I call ```FastLED.addLeds``` more than 7 times on my ESP-12E based NodeMCU. For clarity, I'm using FastLED v3.1.6, esp8266 board v2.4.0, Arduino IDE v1.8.5, Mac OSX v10.13.3 -- which is the latest stable version for each piece of software.

Until this is reworked, could the documentation reflect this limitation? I built a 9 strip chandelier and to avoid putting in a 4th wire, I decided to use extra pins on the NodeMCU. Now I'll have to tie 3 strips together to not exceed 7 strips. Thanks for all your work on this library!

kuike-tiel, 2019-02-27:
Any updates on this??

I have something like this (only on a esp8266), i am trying to make a led controller with the fastled library where the user can input wich led type they have and wich RGB arrangnement the led has. 
So i did try this and it worked.

```
void setFastled()
{
String lt = ledType;
 if (lt == "WS2812B") {
      ws2812B ();
 }  

void ws2812B ()
{
  String la = ledArrange;
    if (la == "RGB") {
     FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, LED_NUMBER);
    } else if (la == "RBG") {
     FastLED.addLeds<WS2812B, DATA_PIN, RBG>(leds, LED_NUMBER);
    } else if (la == "GRB") {
    FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, LED_NUMBER);
    } else if (la == "GBR") {
    FastLED.addLeds<WS2812B, DATA_PIN, GBR>(leds, LED_NUMBER);
    } else if (la == "BRG") {
     FastLED.addLeds<WS2812B, DATA_PIN, BRG>(leds, LED_NUMBER);
    } else if (la == "BGR") {
     FastLED.addLeds<WS2812B, DATA_PIN, BGR>(leds, LED_NUMBER);
    }
    FastLED.setMaxPowerInVoltsAndMilliamps(5, LED_LIMIT_MILLIAMPS);
  }

```

But when i do add a few more.
```
void setFastled()
{
String lt = ledType;
 
 if (lt == "WS2812B") {
      ws2812 ();
 }  
    if (lt == "WS2812") {
      ws2812 ();
    }
    if (lt == "WS2811") {
      ws2811 ();
    }
    if (lt == "WS2811_400") {
      ws2811_400 ();
    }
    if (lt == "SM16703") {
      sm16703 ();
    }
    if (lt == "TM1829") {
      tm1829 ();
    }
    if (lt == "TM1812") {
      tm1812 ();
    }
    if (lt == "TM1809") {
      tm1809 ();
    }
    if (lt == "TM1804") {
      tm1804 ();
    }
  }

void ws2812B ()
[... 151 lines cut ...]
 String la = ledArrange;
    if (la == "RGB") {
      FastLED.addLeds<TM1804, DATA_PIN, RGB>(leds, LED_NUMBER);
    } else if (la == "RBG") {
      FastLED.addLeds<TM1804, DATA_PIN, RBG>(leds, LED_NUMBER);
    } else if (la == "GRB") {
      FastLED.addLeds<TM1804, DATA_PIN, GRB>(leds, LED_NUMBER);
    } else if (la == "GBR") {
      FastLED.addLeds<TM1804, DATA_PIN, GBR>(leds, LED_NUMBER);
    } else if (la == "BRG") {
      FastLED.addLeds<TM1804, DATA_PIN, BRG>(leds, LED_NUMBER);
    } else if (la == "BGR") {
      FastLED.addLeds<TM1804, DATA_PIN, BGR>(leds, LED_NUMBER);
    }
    FastLED.setMaxPowerInVoltsAndMilliamps(12, LED_LIMIT_MILLIAMPS);
  }

```

i get this error message:  

   section `.text' will not fit in region `iram1_0_seg'

 I am a noob and this is my first coding/project. So i do not know what i am doing wrong here. 
I tried this with a brand new sketch and also got the same error so it is not that my sketch is to big (my geuss).

focalintent, 2019-02-27:
Please stop spamming every multi-controller issue with this question.  FastLED is not designed for building a generic controller like you are trying to - separate code gets compiled for every combination of controller, pin, and rgb ordering.  Which is why you are running out of program space.  Little short of an entire library rewrite will allow for this (and I haven’t decided yet whether or not I want go piggyback that onto the major rewrite that I’m doing for rgb/16-bit support)

kriegsman, 2020-02-17:
Cleaning up old issues; closing this one.

closed 2020-02-17 by kriegsman