FastLED/FastLED #419: Teensy 3.1/3.2 ARMHardwareSPIOutput::waitFully hanging due to race condition

mikeburg, 2017-03-05:
I've found a race condition that causes ARMHardwareSPIOutput::waitFully to hang while high frequency asynchronous operations (aka interrupts) are happening.

This could very well be related to issue #339 and the associated fix to try and correct it.

The project I'm working on is LED 64x32 APA102-based matrix with a microphone attached to a Teensy 3.2 board. The teensy's Audio library is being used to do FFT & peak analysis for an animated spectrum displayed on the matrix.

The problem I kept encountering was the code would hang randomly some 30 seconds to 10 minutes after booting. After several hours of ruling out my code and the Teensy Audio library as the problem, it turns out FastLED.show was never returning. Digging deeper, ARMHardwareSPIOutput::waitFully was looping forever waiting SPI_SR_TCF (transmit complete) to be asserted which never happened.

What's happening is there is a window of opportunity for things to go wrong between when writeByte* pushes a byte to the SPI FIFO, and when the TCF bit is cleared. The scenario is the byte is pushed, the audio DMA interrupt triggers and is serviced, the audio data is crunched (via Software_IRQ), and then writeByte is resumed where it clears the TCF bit. During this time, its very possible the SPI module has transmitted the byte across the bus and set the TCF bit to indicate completion. So when the writeByte routine resumes, the TCF bit is cleared erroneously for the byte just pushed which in turns causes waitFully to hang because it is expecting the last byte transmitted to have TCF set.

The solution I came up with is:

1. Lock out interrupts in writeByte* to protect the byte push and bit clear from interrupt and high interrupt latency.  That clears up the problem.
2. As an optimization, switch over to using EOQF (End of Queue Flag) instead of TCF. By using EOQF the waitFully routine only needs to busy wait for the EOQF bit to be asserted without having to disable interrupts in order to worry about clearing any existing TCF flags while the fifo is not empty. Two busy loops are reduced to one. In writeByte*, EOQF is cleared in place of TCF.
 
I can make a push request with the fix if this all makes sense. The same fix should apply to Teensy 3.5/3.6 (K66) but I do not have a way to test against those boards right now.

-- Michael

mikeburg, 2017-03-05:
Ugh, strike that EOQF optimization. I misread the datasheet on its meaning. I would still change the original busy wait to disable interrupts read the fifo count, clear TCF if non-zero, re-enable interrupts then loop so other hardware is not locked out.

RobinB, 2017-03-28:
Hi Mikeburg! I'm encountering this exact problem with my project (a one-dimensional dungeon crawler called Line Wobbler). Could you post a quick-fix somewhere? I was getting really desperate and even considered using two teensys talking over serial ...

RobinB, 2017-03-28:
Actually, one of the very recent fixes for FastLED (on the same day you noticed this) fixes this problem. I lost so much time on this, glad to see it's fixed!

mikeburg, 2017-03-28:
Hi RobinB - I submitted a pull request which was accepted ~3 weeks ago that fixes this.

The fix also applies to the Teensy 3.5 & 3.6 boards except I haven't had time to test and submit a pull request.

florianlink, 2017-05-14:
Thanks mikeburg, this fixes my hangs that I experienced with FastLED + FFT + Serial1 on Teensy 3.2!
I think the bug can be closed.

commit 6f445905 (2017-08-16): Pull the SPI race condition fix over to the teensy 3.6 spi code as well.  Fixes #419 for teensy 3.6 as well
closed 2017-08-16 by focalintent