FastLED/FastLED #421: Fireduino compilation error

samuei, 2017-03-08:
"Help! I'm getting a compiler error about "avr/io.h" not being found!"

I kickstarted the Fireduino board ([SDK](https://github.com/FireflyTeam/Fireduino-SDK) / [specs](http://wiki.t-firefly.com/index.php/Fireduino/Fireduino_Hardware/en
)) a while back, and was trying to use FastLED with it, but I'm getting an error about avr/io.h not existing. Your FAQs suggest that means it's not compatible, but I'm really hoping there's an easy workaround. It's an ARM Cortex-M3 and it uses the Arduino software, so here's hoping.

focalintent, 2017-03-08:
Because FastLED talks directly to low level I/O registers for pin control, it requires porting work to run on a new hardware platform which this is.  

So, no easy workarounds really :/ - I can add this to the list of future boards to support.

samuei, 2017-03-08:
Well, it was worth a shot.

focalintent, 2017-03-08:
I'll add it to the list of things to investigate support for - but major library work is on hiatus at the moment, so I can't promise any timing.

kriegsman, 2020-02-17:
Cleaning up old issues; closing this one.

closed 2020-02-17 by kriegsman