FastLED/FastLED #423: FastLED without AVR

Atraii, 2017-03-12:
This question feels in bad taste, but I am too curious to leave it alone. 

Is it possible to use FastLED without any output libraries on an x86 computer? How bad of an idea would this be? 

Background: I was originally hosting a simple server and a fastled implementation on ESP8266. Due to memory limitations, I found that I couldn't accomplish all that I wanted to. Instead, I would like to simply use the ESP8266's to receive color information via UDP and display that on the strips. I have tested the basic idea for the script, and it works rather well (900 leds across 2 ESP8266s, 120 fps, with no lag as long as it is on a dedicated access point). 

However, now I find myself without all the great functions I used from FastLED (CHSV structs, fast sin implementations, fill functions, nscale8, etc...). Is it reasonable/possible to still access and use all of the amazing helper functions, or am I better off re-implementing what I need in C++ as I go?

Atraii, 2017-03-13:
Although it doesn't address it directly, I just found a [recent pull request](https://github.com/FastLED/FastLED/pull/422) that adds linux platform support and allows the full FastLED library to compile under x86. 

Is the best way to go about using the FastLED library in a  "headless" configuration to use this or utilize a dummy platform?

focalintent, 2019-01-20:
There will be a "null" platform being added to the library soon which will take care of this.

closed 2019-01-20 by focalintent