FastLED/FastLED #424: WS2812B just turns on white

usersxm, 2017-03-13:
I've tried this library, and the adafruit one but no one seems to be working for me. I don't know what's the problem, I've also tried using an arduino uno and a nano to power just one led (cut off from a led strip).
Here it is the sketch for the blinking example, and also a picture of all the wirings (in case you cannot see, there's a 100ohm resistor for the data pin which I've read was recommended):
[image]
```
#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 1

// For led chips like Neopixels, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
#define DATA_PIN 5
//#define CLOCK_PIN 13

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
      // Uncomment/edit one of the following lines for your leds arrangement.
      FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

The most weird thing is that when I unplug the arduino from the computer and plug it back again the led won't turn on, I need to move, unplug the 5v, ground and data pins and sometimes after doing this it turns back on. It is driving me mad! How is it possible that I can't achieve such a simple task!

focalintent, 2017-03-13:
See the arrow on the strip?  That indicates which way data is flowing.  You've wired things to the output side of the led (DO == data out) instead of the input. Move your connections to the other side of the led chip.

closed 2017-03-13 by focalintent

focalintent, 2017-03-13:
Nothing that I can obviously see here - but I prefer to keep GitHub issues for bug reports a if you ask on the g+ community (http://FastLED.io/+) you will be more likely to come across folks who can definitively answer you.

electrokean, 2017-03-13:
@usersxm it looks like you don't have anything connected to the -V side of the power supply, but instead you're using the earth terminal.  The earth is for mains earthing for safety, and is not necessarily common with the -V (usually there is a capacitor between them, so no current flows).  Also make sure you have a common ground connection between the LEDs, PSU (-V), and Arduino.