FastLED/FastLED #425: Not possible to declare ports as variables because of template

dimitre, 2017-03-13:
I would like to add new led strips from an array or struct but it seems the template system doesn't recognize variables as parameters.
This code reproduces the issue. 
Thanks

```c++
#include "FastLED.h"

#define NUM_LEDS 80
float contagem = 0;
CRGB leds[NUM_LEDS];

void setup() {
  int d = 7;
  int c = 14;
  LEDS.addLeds<APA102, d, c, BGR, DATA_RATE_MHZ(10)>(leds, NUM_LEDS);
}
```

NicoHood, 2017-03-13:
use `#define` instead. or `const int` might also work

dimitre, 2017-03-13:
Thanks @nicohood, in fact I'm trying this:
Any ideas?

```c++
struct port {
public:
	uint8_t data;
	uint8_t clock;
	port (uint8_t d, uint8_t c) : data(d), clock(c) {};
};

port ports[] = { 
	port(0, 1),
	port(2, 3),
	port(4, 5),
	port(6, 7),
	port(14, 15),
	port(16, 17),
	port(18, 19),
	port(20, 21),
	port(22, 23),
	port(8, 9)
};

for (auto & p : ports) {
	LEDS.addLeds<APA102, p.data, p.clock, BGR, DATA_RATE_MHZ(10)>(leds, NUM_LEDS); //, BGR
}
```

kcouck, 2017-03-13:
Did you try declaring d and c as uint8_t instead of int?
Do you get a compile error?

And if you're not going to change those values program, a define should work as well...

NicoHood, 2017-03-13:
As i said, try `const`

focalintent, 2017-03-13:
The template parameters have to be constants - there's really no way around this one without major re-working of the library internals (and I use the constants to, on a number of platforms, determine GPIO/mask information at compile time which allows for more efficient GPIO operations (1-2 cycles vs. the multiple cycles having them in variables would take vs. the ~100+ cycles that digitalWrite takes).

closed 2017-03-13 by focalintent