FastLED/FastLED #426: Teensy 3.2 strange behaviour / bugs

DaKoalP, 2017-03-16:
Hey there!
I'm currently working on a project involving about 1000 WS2812b LEDs and wanted to control them using the Teensy 3.2 with a self-made Octows2811 adapter. Everything was working fine, until I started programming…
I’ve encountered these 3 problems:
1.)	The teensy will crash whenever I try to write out RGB values with a higher number of leds than I actually have (for instance: I have 100 leds on 1 output port and try to write “CRGB::Red” to led 101). This is not a huge problem, since I can, of course, simply avoid writing to these, but it makes programming quite toilsome… Also: writing “CRGB::White” doesn’t cause that mistake.
2.)	This is the tricky one: Cycling all of the leds through the colour space using the HSV/hue commands works like a charm, however, cycling only a few of them will cause them to turn darker and darker until they are completely black. When they get back to red, they slowly turn on again and reach, at least I think, full brightness. At first I thought this had something to do with colour correction, but then again it shouldn’t work with all leds at the same time then, right? 
3.)	When using the FastLED.delay function, the animation will randomly get slower. I’m not really sure about this though, it might just be my perception. I can say for sure though, that using the normal delay function speeds up the whole animation.
If it helps, I can upload a video showing the 2nd problem. 

Here is some basic code that shows the problem:

Setup:

```
LEDS.addLeds<NEOPIXEL, 6>  (leds[0], NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
LEDS.addLeds<NEOPIXEL, 20> (leds[1], NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
LEDS.addLeds<NEOPIXEL, 21> (leds[2], NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
LEDS.addLeds<NEOPIXEL, 5>  (leds[3], NUM_LEDS_PER_STRIP).setCorrection(TypicalLEDStrip);
LEDS.setBrightness(255);
```

Loop:

```
  uint8_t pixel1 = 0;
  uint8_t hue = 0;

    LEDS.clear();
    for(int i = 0; i < NUM_STRIPS; i++)
    {
      leds[i][pixel1] = CHSV(hue, 255, 255);
      leds[i][pixel1+1] = CHSV(hue, 255, 255);
    }
  
    LEDS.show();
    delay(60);
    
    pixel1++;
    hue++;
    
    if(pixel1 >= NUM_LEDS_PER_STRIP) pixel1 = 0;
```

As soon as `pixel1 == NUM_LEDS_PER_STRIP` the teensy gets stuck, since I'm trying to write to a led that doesn't exist (`leds[i][pixel1+1] = CHSV(hue, 255, 255);`).

kcouck, 2017-03-16:
1. The Teensy probably crashes because you're writing to unallocated memory (outside of your array). You're not writing out leds to your strips yet, that's what LEDS.show() does.
2. check with putting RGB values (for red, green, blue) if the strip has the correct RGB byte order...
3. delay() does a show() as well, but after waiting a certain time. So you're writing out your leds twice if you do LEDS.show()+ LEDS.delay()

Random performance remark: if you're wring out 2 pixels at a time and extinguishing the previous one, remove the LEDS.clear() and just write out RGB::Black to leds[i][pixel1-1].
To solve 1) and avoid issues with this remark, put an 'if' onthose writes to check your array bounds:
```
if (pixel1 > 1) leds[i][pixel1-1] =  CRGB::Black;
leds[i][pixel1] = CHSV(hue, 255, 255); 
if (pixel1 < NUM_LEDS_PER_STRIP-1) leds[i][pixel1+1] = CHSV(hue, 255, 255);
```

DaKoalP, 2017-03-16:
1.) That's what I originally thought, but since it works with white I didn't think that this would be the cause. I just tried it with a larger array and it actually works! What a dumb mistake, thanks man! :)
2.) Increasing the array size also solved this problem, I don't get why though.
3.) That does also make a lot of sense, thanks again!

I also tried said solution with checking the array boundries, interestingly the same problem with the brightness of the leds occured again. I guess I'll just stick with the larger array size, since increasing it by 5 won't decrease performance when we're talking about 250 leds per output.

So the question why the leds get darker stays, but generally the mistery can be marked as solved.

Gotta say thank you one more time 🥇

kcouck, 2017-03-16:
I don't see how the array size may influence your second issue, might be something else in the code though...

As I said, try writing RGB::Blue, RGB::Red and RGB::Green and see if the led colors are correct.
I see you're creating them as Neopixel strips, they are GRB; the default value for others is RGB... But I guess you'd just see green showing up instead of red in your example...

Another thing; I remember seeing somewhere that driving the leds at maximum brightness might overheat them after a while, dropping out chunks of led strip. Your 'dimming and going to red' remark makes me wonder if it isn't a power issue... Try bringing the brightness down to a lower value? 128 (50%) for example?

Good luck :)

focalintent, 2017-03-16:
The effects of writing outside of an array can be random.  It depends on what is occupying the memory you are writing into.  Maybe 255 is the value that was supposed to be in what you were writing into, which is why white "works".

DaKoalP, 2017-03-17:
> As I said, try writing RGB::Blue, RGB::Red and RGB::Green and see if the led colors are correct.
I see you're creating them as Neopixel strips, they are GRB; the default value for others is RGB... But I guess you'd just see green showing up instead of red in your example...

No, the colors are correct.

> Another thing; I remember seeing somewhere that driving the leds at maximum brightness might overheat them after a while, dropping out chunks of led strip. Your 'dimming and going to red' remark [...]

Unfortunately also no, changing the brightness didn't show an effect. Wonder what the cause is then... Basically the code example I showed you is exactly what's in my program so I guess it's not a code-related issue?

>The effects of writing outside of an array can be random. It depends on what is occupying the memory you are writing into. Maybe 255 is the value that was supposed to be in what you were writing into, which is why white "works".

That would make sense! Thanks for the tip :)

closed 2017-03-19 by DaKoalP