FastLED/FastLED #427: APA102 requires show() to be called twice for a correct update

GentleGiantSoftware, 2017-03-16:
The following code demonstrates the issue I am seeing with FastLED version 3.0001.003. When running this code for a WS2811 it works fine but when using the APA102 it requires FasltLED.show() to be called twice to update the leds correctly.

#include <FastLED.h>
int numLeds = 144;
CRGB* leds;
  
void setup() {
    Serial.begin(115200);
    leds = (CRGB*)malloc(numLeds * sizeof(CRGB));
    FastLED.addLeds<APA102, 24, 25, BGR>(leds, numLeds); 
}

void loop() {
  Serial.println("White");
  for (uint16_t i = 0; i < numLeds; i++)   {
    leds[i] = CRGB::White;
  }
  FastLED.show(); // ==> Leds are black for APA102, white for WS2811 as it should
  delay(1000);

  Serial.println("Black");
  for (uint16_t i = 0; i < numLeds; i++)   {
    leds[i] = CRGB::Black;
  }
  FastLED.show(); // ==> Leds are white for APA102, black for WS2811 as it should
  delay(1000);
}

When you see "White" the leds will be off while when you see "Black" the leds will be on, which is not expected. When I use the WS2811 leds it works fine as does it when I call FastLED.show(); twice which should not be needed. 

Regards,
Ben

focalintent, 2017-03-16:
this is because apa102's don't latch new colors in until they see the start frame.  i think i put in a fix for this already - grab the latest out of github.

GentleGiantSoftware, 2017-03-17:
Thanks for the quick response, just pulled the latest from GitHub which shows version 3.001.005 during the build. Unfortunately the behavior has not changed. Still need to call FastLED.show(); twice before the leds will show the defined colors.

focalintent, 2017-03-17:
Actually try setting your led type to SK9822 - the behavior you're describing sounds more like that (and I've heard reports of folks being shipped the SK9822 when ordering APA102)

GentleGiantSoftware, 2017-03-18:
thank you, this also seems to be the issue for me, when I selected the sk9822 as led type it works fine.

closed 2017-03-28 by focalintent