FastLED/FastLED #429: Hardware SPI for ESP8266?

rlesac, 2017-03-27:
It seems that FastLED is not using hardware SPI on ESP8266.
I define the right pins, GPIO14 for clock and GPIO13 for data, but get a compile time warning that "No hardware SPI pins defined.  All SPI access will default to bitbanged output".

I'm using ESP-12e, with raw pin order.

Here is a relevant page discussing esp8266 and HSPI.

focalintent, 2017-03-27:
All work on the library is currently suspended.  Hardware SPI for the esp8266 is on the list for when I get back to working on it.

Leozolt, 2017-11-26:
Is it possible to motivate development on the hardware SPI for the ESP8266?

focalintent, 2017-11-26:
No, I haven’t gotten back to a place of working on the library yet.

Leozolt, 2017-11-26:
Do you accept donations to ease (facility estate assets and) time on this project? Sorry if this is off-topic.

coryking, 2017-12-12:
Just FYI, I've added support for DMA on [my own personal fork](/coryking/FastLED).  Works for ESP8266 only and is so far buttery smooth....

the-magister, 2018-01-13:
@focalintent https://github.com/Makuna/NeoPixelBus/blob/master/src/internal/NeoEsp8266UartMethod.h provides either UART or SPI-based support for ESP8266.  I took a browse through the code, and it's too heady for me to understand how compatible it is.  To me, the loss of Serial1 (which is very seldom used) via the UART-based approach seems reasonable.

Essentially, someone driving a ton of lights should be using the PORT approach or go over to a Teensy.  Someone driving a strand of lights and wanting to use WiFi would benefit from the UART or DMA approach.

nsepehr, 2018-07-11:
I would love to see the SPI support for the ESP8266. When you pass the 1000 LED mark it makes a huge difference having the SPI support. Thanks for all of the hard work you put into this library. It's the fastest I've seen around. Hope you can come back to it for more upgrades.

ctrowat, 2019-01-10:
I really don't mean to be insensitive but I see some time has passed on this issue and I was hoping I could ask for an update.  I am considering using a large number of APA102 with an ESP8266 on a current project.  Has there been any progress made on this particular issue, or are there any other solutions to use hardware SPI in this configuration?  I really like your library and want to use it everywhere I possibly can.

focalintent, 2019-01-20:
No one has submitted code to me for the hardware SPI and while I'm starting back in on library work, I'm focusing mostly on the rewrite for RGBW/16-bit output - after which I'll go back to adding hardware platforms/features support.

vayulove, 2019-03-16:
> Just FYI, I've added support for DMA on [my own personal fork](/coryking/FastLED). Works for ESP8266 only and is so far buttery smooth....

You mention to use GPIO 3. Will this work for a strip that needs clock and data?  FastLED is really really slow on my nodemcu. I've been using 14 for clock and 13 for data. Would that fork help? What pins for clock and for data would I use for my nodemcu?

BenoitAnastay, 2019-12-11:
That should work

https://github.com/FastLED/FastLED/pull/936

dbu, 2023-11-25:
this issue can be closed, it was implemented in #936

`#define FASTLED_ALL_PINS_HARDWARE_SPI` before `#include "FastLED.h"` to activate it