FastLED/FastLED #431: FastLED compilation issue with Eclipse and arduino

SigP226, 2017-03-28:
Hi, 

I becomes crazy since couple of hours. impossible to compile FastLed with Eclipse for Arduino.
Configuration : Eclipse, CDT, AVR plugins and configuration done for Arduino. I can build easily an "Hello world" program and download it in UNO R3 or Mega 2560 board. Even a FreeRTOS sample code works fine, so compiler option and procedure are OK.

I use Fastled library version 3.001.003 which works fine in Arduino IDE (1.8)

But as soon as I try to add FastLed source in my project under Eclispe and build it...impossible. I get many errors during build.

The long list of errors start with this one
FastLED/controller.h: In constructor 'CLEDController::CLEDController()':
FastLED/controller.h:64:31: error: 'NULL' was not declared in this scope

I think this error is due to lack of arduino board type configuration. 
I have tried to use include"arduino.h" in FastLED.H but it doesn't solve completely the issue. I get other errors related to the lack of "avr" definition (it doesn't take the good sysdef.h include in avr folder) like this:

"In file included from C:/Users/Bricoleur/Documents/Arduino/libraries/FastLED/FastLED.h:49:0,
                 from C:/Users/Bricoleur/Documents/Arduino/libraries/FastLED/FastLED.cpp:2:
C:/Users/Bricoleur/Documents/Arduino/libraries/FastLED/fastpin.h:34:11: error: 'RwReg' does not name a type"

RwReg is defined in ledsysdef_avr.h, so it means, it doesn't find the arduino board configuration.

Any ideas are welcome
Thanks for your help

focalintent, 2017-03-28:
Right now I only support building FastLED inside of the arduino ide - and haven't done the work to support building underneath eclipse (and I'm currently not doing any major work on the library right now).

My recommendation would be to ask on the g+ group - http://fastled.io/+/ - and see if there is anyone there who has done the work to get the project building under eclipse.  FastLED does a lot of low level access of register i/o, and to be able to do that properly there are chip/system definitions that have to be set correctly so the right choices get made by the compiler.

closed 2017-03-28 by focalintent

SigP226, 2017-03-29:
Thanks a lot for your answer. I am going to ask to g+ group.

SigP226, 2017-03-29:
SOLUTION FOUND! this is due to preprocessor option defined during arduino IDE compilation job which are missing. -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR for my project.
first is the IDE version, second is generated by preprocessor and depend of the board, third also.