FastLED/FastLED #432: Maximum number of LEDs with a Teensy3.2?

mgcrea, 2017-03-31:
It looks like I'm hitting an unexpected hard limit with the FastLED library, while trying to drive 2960 leds. The sketch does not start. The teensy 3.2 having 64K RAM I don't think it is the issue (2960*3

I've got basically the same sketch using directly the `octows2811` library and it works:

```
#define NUM_LEDS_PER_STRIP      (74 * 5)
#define NUM_STRIPS              8
#define NUM_LEDS                (NUM_STRIPS * NUM_LEDS_PER_STRIP)
```

It does work with less leds:

```
#define NUM_LEDS_PER_STRIP      (60 * 5)
#define NUM_STRIPS              8
#define NUM_LEDS                (NUM_STRIPS * NUM_LEDS_PER_STRIP)
``` 

Is it a library bug or am I hitting a hard limit?

Thanks!

Files used to test:

- octows2811.c:

```
#include "./octows2811.h"

namespace octows2811 {

  void colorWipe(int color) {
    for (int i=0; i < NUM_LEDS; i++) {
      leds.setPixel(i, color);
    }
    leds.show();
  }

  void colorWipe(int color, int wait) {
    for (int i=0; i < NUM_LEDS; i++) {
      leds.setPixel(i, color);
      leds.show();
      delayMicroseconds(0);
    }
  }

  void setup() {
    leds.begin();

    colorWipe(RED);
    delay(500);
    colorWipe(BLACK);
    delay(500);
  }

  void loop() {
    int microsec = TEMPO_MICROSEC / NUM_LEDS;  // change them all in TEMPO_MICROSEC seconds

    colorWipe(RED, microsec);
    colorWipe(GREEN, microsec);
    colorWipe(BLUE, microsec);
    colorWipe(YELLOW, microsec);
    colorWipe(PINK, microsec);
    colorWipe(ORANGE, microsec);
    colorWipe(WHITE, microsec);
    colorWipe(0xFFFFFF, microsec);
  }

} // namespace octows2811
```

- fastled.c:

```
#include "./fastled.h"

namespace fastled {

  void colorWipe(int color) {
    for (int i = 0; i < NUM_LEDS; i++) {
      leds[i] = color;
    }
    FastLED.show();
  }

  void colorWipe(int color, int wait) {
    for (int i = 0; i < NUM_LEDS; i++) {
      leds[i] = color;
      FastLED.show();
      delayMicroseconds(wait);
    }
  }

  void setup() {
    FastLED.addLeds<OCTOWS2811>(leds, NUM_LEDS_PER_STRIP);
    // FastLED.setBrightness(LED_BRIGHTNESS);
    colorWipe(RED);
    delay(500);
    colorWipe(BLACK);
    delay(500);
  }

  void loop() {
    int microsec = TEMPO_MICROSEC / NUM_LEDS;  // change them all in TEMPO_MICROSEC seconds

    colorWipe(RED, microsec);
    colorWipe(GREEN, microsec);
    colorWipe(BLUE, microsec);
    colorWipe(YELLOW, microsec);
    colorWipe(PINK, microsec);
    colorWipe(ORANGE, microsec);
    colorWipe(WHITE, microsec);
    // colorWipe(0xFFFFFF, microsec);
  }

} // namespace fastled
```

focalintent, 2017-03-31:
How're you defining leds?  Also - because of how FastLED does its double buffering to talk to OctoWS2811 - it's going to use 26640 bytes just for the led data - your led buffer and the two internally - it's possible that you're running into a limit with the dynamic memory allocator, which may not be easily worked around, I'll have to do some testing when I get back to some hardware.  

(Also - what you've supplied isn't really a complete buildable test (I don't have a whole lot of time to put into FastLED these days (the project is still on indefinite hiatus/hold) - not having to guess at what else you've got missing would be helpful) - so if I do some quick tests building my own test app and it works - then I'm stalled waiting for you to showe me the rest of your code/setup in case there's something in there exacerbating things))

focalintent, 2017-03-31:
Also - if your code is building for both OctoWS2811 directly _and_ FastLED at the same time (which is my guess given how you're name spacing) - then that actually ups your baseline memory usage to 45k - but even more than that, if you're marking some of the memory for OctoWS2811 as DMAMEM that's going to shuffle how ram is laid out in a way that might also impact the malloc calls in FastLED (welcome to the wonderful world of memory management and fragmentation)

mgcrea, 2017-03-31:
@focalintent I had three headers files loaded (octows, a fork octosk and fastled) and having static DMAMEM calls in these headers was indeed the issue! Not sure how I missed that! Thanks!

closed 2017-03-31 by mgcrea

focalintent, 2017-03-31:
Also -  I have some suspicions about how you're building your project - and I bet you're basically doubling your memory usage needs by having a set of memory for using octows2811 directly and a set of memory for using fastled w/the octows2811 driver.  So - basically, 5x the number of leds (vs. 2x for octows2811 direct and 3x for fastled w/octows2811 driver)

mgcrea, 2017-03-31:
@focalintent I'm using [platform.io](http://platformio.org/) to build the project, here is a rough layout of the files:

```
$ tree -L 3
.
├── Makefile
├── README.md
├── lib
│   ├── FastLED
│   │   ├── FastLED.cpp
│   │   ├── FastLED.h
│   │   ├── LICENSE
│   │   ├── PORTING.md
│   │   ├── [...]
│   ├── OctoSK6812
│   │   ├── OctoSK6812.cpp
│   │   ├── OctoSK6812.h
│   │   ├── README.md
│   │   ├── examples
│   │   ├── library.json
│   │   └── library.properties
│   ├── OctoWS2811
│   │   ├── OctoWS2811.cpp
│   │   ├── OctoWS2811.h
│   │   ├── README.md
│   │   ├── examples
│   │   ├── extras
│   │   ├── keywords.txt
│   │   ├── library.json
│   │   └── library.properties
│   └── readme.txt
├── platformio.ini
└── src
    ├── app.cpp
    ├── config
    │   └── config.h
    ├── led
    │   ├── effects.cpp
    │   ├── effects.h
    │   ├── led.cpp
    │   └── led.h
    ├── tests
    │   ├── fastled.cpp
    │   ├── fastled.h
    │   ├── legacy
    │   ├── octows2811.cpp
    │   └── octows2811.h
    └── utils
```

- `app.cpp`:
```
#include "config/config.h"

// #include "./led/led.h"
// #include "./tests/octows2811.h"
#include "./tests/fastled.h"

unsigned long count;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  pinMode(A3, INPUT);
  pinMode(A8, INPUT);
  pinMode(A9, INPUT);
  pinMode(10, INPUT_PULLUP);

  // led::setup();
  // octows2811::setup();
  // octosk6812::setup();
  fastled::setup();

  digitalWrite(LED_BUILTIN, LOW);
}

void loop() {
  // led::loop();
  // octows2811::loop();
  // octosk6812::loop();
  fastled::loop();

  digitalWrite(LED_BUILTIN, count % 4 ? HIGH : LOW);
  count++;
}
```

- `src/tests/fastled.h`:

```
#ifndef fastled_h
#define fastled_h

#include "config/config.h"

#define USE_OCTOWS2811
#include <OctoWS2811.h>
#include <FastLED.h>

// FASTLED_USING_NAMESPACE
// #if defined(FASTLED_VERSION) && (FASTLED_VERSION < 3001000)
// #warning "Requires FastLED 3.1 or later; check github for latest code."
// #endif

namespace fastled {

  static CRGB leds[NUM_STRIPS * NUM_LEDS_PER_STRIP];
  static uint8_t gHue = 0; // rotating "base color" used by many of the patterns

  void setup();
  void loop();

}

#endif // fastled_h
```

- `src/tests/octows2811.h`:

```
#ifndef octows2811_h
#define octows2811_h

#include "config/config.h"
#include <OctoWS2811.h>

namespace octows2811 {

  static DMAMEM int displayMemory[NUM_LEDS_PER_STRIP * 6];
  static int drawingMemory[NUM_LEDS_PER_STRIP * 6];
  static const int config = WS2811_GRB | WS2811_800kHz;

  static OctoWS2811 leds(NUM_LEDS_PER_STRIP, displayMemory, drawingMemory, config);
  static uint8_t gHue = 0; // rotating "base color" used by many of the patterns

  void setup();
  void loop();

}

#endif // octows2811_h
```

- `platformio.ini`:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:teensy31]
platform = teensy
board = teensy31
framework = arduino
```