FastLED/FastLED #433: XYsafe is supposed to return -1 but is defined as uint16_t

rhulha, 2017-04-01:
Hi All,

I noticed that XYsafe returns 65535 instead of -1 as is needed to adres the safe pixel.

The reason is that XYsafe is defined to return uint16_t instead of a signed integer.
```
uint16_t XYsafe( uint8_t x, uint8_t y)
{
  if( x >= kMatrixWidth) return -1;
  if( y >= kMatrixHeight) return -1;
  return XY(x,y);
}

```

kriegsman, 2017-04-01:
Thanks for raising the question.  

I'm 99% sure that the existing code will do exactly the correct thing, as written.  In C, with 16-bit values, adding 65535 is literally the same as subtracting one.  Here's an example:

```
  uint16_t i;
  i = 10;
  i = i + 65535;  
  // numerically, 10 + 65535 = 65545
  // but 65545 overflows past the limit of what fits in 16 bits (0..65535)
  // so what's put into i is essentially the low 16 bits of 65545,
  // which is the same as (65545 & 0xFFFF), which is ... 9
  // so now i = 9, 
  // and we see that the effect of adding 65535 is identical to subtracting 1
```
Another way to think about it is that the binary representation for -1 is 0xFFFF (in a 16-bit world), which is literally the same as the representation for 65535.  So while you are correct, that the function is returning 0xFFFF, using it as an array offset has the same effect as subtracting one.

In a 32-bit environment, I could see that there might be some problem if the compiler isn't sign-extending the uint16 before adding it to an array offset.  In that case, we'd have to make some changes.

What are you seeing?  Are you running in a 32-bit environment and having a problem with this code presently?

Thanks again for the thoughtful comments, and if you're having problems with your code actually not running correctly in a 32-bit environment, please let me know, and I'll take a look at it.  (Are you running on ARM?  ESP8266? something else?)

rhulha, 2017-04-02:
Ah that makes sense. I am using a plain old Arduino Mega 2560.

So what happened was, that my code didn't work properly. 

So I added a bunch of Serial.println - one being the result of XYsafe.

Now that println gave me 65535 instead of -1 so instead of realizing that the fail safe of XYsafe had triggered I looked at the XY algorithm to find the bug (even used Visual Studio to step through it ) and thus wasted quite a bit of time.

I still don't understand why using 65535 as an array index is the same as -1.
I guess it must be because the compiler casts the index value to an int16_t ?

kriegsman, 2017-04-02:
It's because on AVR-based Arduino systems, all integers default to 16 bits -- unlike in most x86 environments, where they defaults to 32 bits.

So if I have an unsigned array index, "i", and I add 0xFFFF (which can be called 65535 _or_ -1) to it, what I get is a new array index which is one less than the one I started with.

On a 32-bit-integer system, -1 is represented as 0xFFFFFFFF (aka 4294967295).  So on a 32-bit-integer system, if you add 4294967295 to an integer, you get one less than you started out with.

If you're into reading the machine code that the compiler generates (I'm not going to get into that here, but it's interesting and you can pursue it if you wish!), you can actually see that for 8-bit values, the AVR compiler will use "adding 0xFF (255)" as a synonym for "subtracting 1":

For this source code:
```uint8_t i;

void setup() {
  i = 100;
}

void loop() {
  i = i - 1;
}
```
the avr-gcc compiler generates this code for the body of 'loop':
```
 9f ef           ldi     r25, 0xFF       ; 255
 98 0f           add     r25, r24
```
That's basically read as "load the number 0xFF (aka 255) into register r25, and then ADD that value to register r24 (which is holding the variable i)".

Wikipedia has more about "two's compliment" if you want to wade into that end of it, too.  https://en.wikipedia.org/wiki/Two%27s_complement  But basically, 0xFFFF (65535) is exactly the same number as 0xFFFF (-1) in a sixteen-bit universe.

Closing the ticket.  For further discussion, check out the FastLED G+ group, which has many more people who can probably offer more guidance!  http://fastled.io/+

eltang, 2017-04-02:
I do believe that there is a potential for this code not to work properly on 32-bit systems. The solution should be to change the return type of XYsafe() to int. The return type of XY() can also be changed, but that shouldn't be strictly necessary.

kriegsman, 2017-04-02:
I think you may be correct, which is why I was wondering if you were hitting actual observed problems. Depends on what gets Sign-extended. But I suspect it might have a bug there. I'll look at it when I get a chance, on 32 bit systems.

kriegsman, 2017-04-02:
Reopening as a reminder.

eltang, 2017-04-02:
I believe that no sign-extension is taking place in this code. A uint16_t is an unsigned type, meaning that there's simply no sign to extend. The 0xFFFF returned is interpreted as 65535.

kriegsman, 2017-04-02:
Yes, but the net result depends on the type (width) of the array index. Nonetheless, this could be better.