FastLED/FastLED #434: FastLED Error

Buzz108, 2017-04-01:
Every FastLED code I try it get this error. Why?

C:\Users\XXX\Documents\Arduino\LED2\LED2.ino:2:21: fatal error: FastLED.h: No such file or directory

 #include "FastLED.h"
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno.

focalintent, 2017-04-01:
What version of Arduino are you using?  Have you downloaded/installed the FastLED library?  Where did you put it?  How did you install it?  Can you provide the full output of the build log (preferably after going into arduino preferences and enabling verbose logging for building)

That error message means that Arduino can't find the library - so it's possible that you've installed it incorrectly.

Buzz108, 2017-04-01:
I figured out where the library is and installed the sketches in there and the now work.

closed 2017-04-02 by focalintent