FastLED/FastLED #435: Teensy v3 re-enables interrupts just long enough for IRRemote ISR to work, thank you!

marcmerlin, 2017-04-04:
So, this is not an issue, but I wanted to open one so that people looking for fastled + IR or IRRemote, hopefully stumble on this.
I don't have permissions to update https://github.com/FastLED/FastLED/wiki/Interrupt-problems 

You indeed cannot run talk to neopixels or 3 wire equivalents and do infrared reading with IRRemote on the same AVR chip because both the adafruit lib and fastled have to disable ISRs to have correct timing when they send the updates to each pixel.
However, the teensy code takes advantage of the fact that the 32bit ARM CPU is fast enough that you can re-enable ISRs for a short time in between pixel updates.
I have verified that this short time period is just long enough for the IRRemote ISR to work.
So, thank you! and if someone with edit access can confirm that things work in https://github.com/FastLED/FastLED/wiki/Interrupt-problems (for teensy v3), that may help others.

jasoncoon, 2017-04-05:
Maybe it could be highlighted or made more obvious, but this is already stated in the wiki article you linked to:

> ... move to the teensy 3.x or the arduino due. These are ARM based systems that have far more clock cycles per second available to them. On these platforms, FastLED will briefly re-enable interrupts between each pixel, to allow handlers to run. As long as those interrupt handlers don't take more than 5µs to run, everything will be happy. As long as your interrupt handlers don't need to run more frequently than once every 30µs, that is.

marcmerlin, 2017-04-05:
I read that, but that doesn't tell me it will actually work with IRRemote, just that it might work with another ISR given that certain conditions are met.
Since the top of the page mentions 'IR' it would be helpful to actually say "yes, it works with IRRemote since its ISR is fast enough on teensy to run within the very short allotted time where interrupts are re-enabled in between pixel updates' (or somesuch)

marcmerlin, 2017-04-17:
This page and video explain the issues between IR and neopixels, and how to make it work on teensy v3.1, ESP8266, and ESP32 (I've now tested all 3 and written code to verify that it works):
http://marc.merlins.org/perso/arduino/post_2017-04-03_Arduino-328P-Uno-Teensy3_1-ESP8266-ESP32-IR-and-Neopixels.html
this short video shows how trying this on AVR is an exercise in frustration and handling failures :)
https://www.youtube.com/watch?v=62-nEJtm070

focalintent, 2017-08-16:
Updated the wiki page with a note about IRRemote on the teensy 3.x

closed 2017-08-16 by focalintent