FastLED/FastLED #436: rgb2hsv_approximate returns completely wrong color

stnkl, 2017-04-06:
Hello,

it feels like I found a bug in rgb2hsv_approximate. For most colors it approximates correctly but for some it is completely wrong.

Sample  code:
```
#include "FastLED.h"

void setup() {
  Serial.begin(115200);
  Serial.println("");
  
  CRGB color1 = CRGB(255,153,0);
  Serial.println("rgb: " + String(color1.red) + "," + String(color1.green) + "," + String(color1.blue));
  CHSV color2 = rgb2hsv_approximate(color1);
  Serial.println("hsv: " + String(color2.hue) + "," + String(color2.sat) + "," + String(color2.val));
}

void loop() {
}
```

Serial output:
```
rgb: 255,153,0
hsv: 122,255,255
```

[image]

Paint.net says rgb(255,153,0) results in hsv(36,100,100) for 360 degrees which would be hsv(26,255,255) for FastLED if I'm not mistaken. Instead rgb2hsv_approximate returns hsv(122,255,255) in FastLED which is not even closely orange-ish (it's green).

The code runs on an ESP8266.

Bug or am I just missing something?

kriegsman, 2017-04-07:
Hi and thanks for the detailed bug report! 
I'll take a look at the code and see if I can reproduce it, and if so get to the bottom of it.

stnkl, 2017-04-07:
Great, thank you! By the way I am using the latest commit (06a5abc) at the moment. If you need anything else let me know!

fabiuz7, 2018-11-12:
Any news about this issue? I'm experiencing the same bug...

kriegsman, 2018-11-12:
Thanks for the reminder. I’ll try to take a look.

fabiuz7, 2018-11-13:
Thank, I was wondering if there are some "weird magics" about this conversion or it is an implementation of formulas describe [here](https://www.rapidtables.com/convert/color/rgb-to-hsv.html) (first google result typing "rgb to hsv"). 

Maybe I could help this issue

burgerga123, 2019-01-12:
I'm also having a similar issue. Trying to convert an RGB palette to HSV to make it easier to intuitively change the colors. Most of them are close enough, but one color that is mostly green comes out completely wrong:
`Original RGB | Converted HSV | Back to RGB
 56   0 108   193 255 204    55   0 108
 48   0 130   187 255 213    50   0 127
 41   0 153   181 255 222    42   0 150
 34   0 175   176 255 231    35   0 173
 34   0 176   176 255 231    35   0 173
 17  44 166   153 190 254    15  43 176
  0  88 156   143 255 249     0  86 157
  0  88 156   143 255 249     0  86 157
  0  94 132   141 255 240     0  90 136
  0 101 108   137 255 231     0 100 109
  0 107  84   134 255 221     0 104  87
  0 108  84   133 255 221     0 108  83
  4 171  68   124 223 255     4 161  69
  9 235  52   146 207 146     2  22  51
  9 235  52   146 207 146     2 22   51
 56   1 108   193 239 210    55   0 108
`

mike2003, 2020-01-26:
Bug - read rgb2hsv_approximate and h=33
h++ and write back to leds[i]
Next read h=33 again!

kriegsman, 2020-01-26:
I’m hoping to make some progress on this this week. Thank you for the examples to reproduce— that’ll help me fix it that much more easily. Much appreciated.

Modifikator, 2020-08-02:
Is there progress in solving this problem?

DrJaymz, 2020-09-11:
This is still broken.  I have used alexa to set the colours and red is red, green is green, blue is blue 
but orange I would say looks like cyan.  Magenta is just blue and yellow is green.
So basically all the secondary colours are way out.

Since alexa only has a few colours I guess I can bodge it but we should get it right.  Is this to do with the hue only have 256 degrees instead of 360?

DrJaymz, 2020-09-11:
roses are red violets are blue and my orange is green and so is my loo.

DrJaymz, 2020-09-11:
actually I found an interesting thing that might help.  I had the LED order wrong so I had GRB.  I then shifted the bytes around so that it looked ok.  This then gives me the right colours, but when you colour fade its wrong.

DrJaymz, 2020-09-12:
here's my temp fix.
 if(r == 255 && g==166) new_colour.h = 21;  //Otherwise it thinks its green which it clearly aint.

5chmidti, 2020-10-01:
I've been looking into this and while I haven't 'fixed' any of the existing functions I've added more color accurate conversion functions in the pr linked above.

More info in the pr, but here are the conversions from this thread rerun with the new functions:
```
r       g       b               h       s       v               r       g       b
56      0       108             192     255     108             54      0       108
48      0       130             185     255     130             43      0       130
41      0       153             181     255     153             37      0       153
34      0       175             178     255     175             30      0       175
34      0       176             178     255     176             30      0       176
17      44      166             163     228     166             18      43      166
0       88      156             147     255     156             0       85      156
0       88      156             147     255     156             0       85      156
0       94      132             140     255     132             0       94      132
0       101     108             131     255     108             0       99      108
0       107     84              118     255     107             0       107     82
0       108     84              118     255     108             0       108     83
0       171     68              101     255     171             0       171     63
0       235     52              94      255     235             0       235     47 // these were problematic 
0       235     52              94      255     235             0       235     47 // but look quite alright here
56      1       108             191     252     108             52      2       108
```

and from the first post:
```
r      g      b        h       s       v
255    143    0        25      255     255
```

cad435, 2022-07-31:
This bug still exists:

`uint32_t tmp = strtol(colorstring.c_str(), NULL, 16);

    Serial.println(tmp);

    CRGB tmpcol = CRGB(tmp);

    Serial.print(tmpcol.r);   
    Serial.print(" ");
    Serial.print(tmpcol.g);
    Serial.print(" ");
    Serial.println(tmpcol.b);

    //task-safe write operation to DotPixelXolor as accessing tasks run on different cores
    xSemaphoreTake(EffektColorAccess, portMAX_DELAY);
    **DotPixelColor = rgb2hsv_approximate(tmpcol); //this approximation somehow does not always work!** 
    xSemaphoreGive(EffektColorAccess);

    Serial.print(DotPixelColor.hue);
    Serial.print(" ");
    Serial.print(DotPixelColor.sat);
    Serial.print(" ");
    Serial.println(DotPixelColor.val);`
    
    
    Console Output:
    
    `Request Col_Set: |#ff8000
    16744448
    255 128 0
    112 255 255`

"Request Col_Set" comes from the calling function from before the code, this is printing purely the string which than is converted int an actual number via strtol();

Color codes are OK until the bold marked line.

 - cad435