FastLED/FastLED #438: spurious warning on ESP8266: No hardware SPI pins defined.  All SPI access will default to bitbanged output

marcmerlin, 2017-04-10:
In file included from /home/merlin/arduino/prg/ardweeny/Neopixel-IR/Neopixel-IR.ino:7:0:
/home/merlin/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /home/merlin/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /home/merlin/arduino/prg/ardweeny/Neopixel-IR/Neopixel-IR.ino:7:
/home/merlin/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Compiling libraries...

Is it expected that I get this warning if I'm not using SPI?

My code:
https://github.com/marcmerlin/Neopixel-IR/blob/master/Neopixel-IR.ino#L375
    FastLED.addLeds<NEOPIXEL,NEOPIXEL_PIN>(leds, NUM_LEDS);

Should the prama be displayed if someone is not using LEDs that use a clock?
When I compile the same exact code on teensy 3.1 or 328p, I don't get the warning.

marcmerlin, 2017-04-10:
In case it's an ESP8266 specific issue (it may be), while we're updating docs (I assume https://github.com/FastLED/FastLED/wiki/ESP8266-notes ), is it possible to add a blurb on interrupts?
I can't quite figure out if the lib uses interrupts or not.
I was able to get https://github.com/markszabo/IRremoteESP826 working at the same time and I didn't expect that it would work. For teensy 3.1 I can see that interrupts are re-enabled to allow such an IR lib to work, but for esp8266 I see no such code, and still, it works :)

focalintent, 2017-04-10:
There isn't a good way for me to tell if someone is trying or not trying to use hardware spi so the warning is always present on platforms that I haven't implemented it on yet.

marcmerlin, 2017-04-10:
@focalintent gotcha. Should this be mentioned on https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging maybe?
And I know it's a somewhat separate issue, but if you happen to know the answer on comment #2 that would be great too (or update the docs to explain the situation too (for interrupts))
Yes, I know writing docs is work, sorry for asking. Normally I'd even send a PR for the first question you just answered,but I don't think I can do that against a wiki I don't have access to, correct?

focalintent, 2017-04-10:
Interrupts should work on the esp8266 - thought that was already documented somewhere - but right now pretty much all work on the library/docs is on hold indefinitely.  I'll leave this open to remind me to update docs when I come back to it.

marcmerlin, 2017-04-10:
@focalintent thanks for the reply. And yes, interrupts seem to work on ESP8266, I guess it works without stopping interrupts at all? (because I don't see it re-enabling interrupts in between pixel updates like teensy3.1 does). Sorry if I was curious :)

focalintent, 2017-04-10:
It does - that's what the os_intr_unlock() and os_intr_lock() calls do.

marcmerlin, 2017-04-10:
ah, great, a 3rd way do cli/sei interrupts/nointerrupts :)
Ok I see it now in clockless_esp8266.h which is probably what I'm running (and it's doing the same re-enable interrupts as teensy3.1 is) while clockless_block_esp8266.h is doing something completely different with interrupts which I'll have to read some more to understand.
Thanks for the pointer

marcmerlin, 2017-04-15:
to avoid mixing up topics, I did more research on IR interrupt on ESP8266 + fastled, and works mostly but glitches a bit. There is a better alternative though: https://github.com/FastLED/FastLED/issues/439

focalintent, 2017-08-16:
This is effectively a duplicate of #429 - hardware SPI for the ESP8266.

closed 2017-08-16 by focalintent