FastLED/FastLED #440: Led stop giving light

Michelvl92, 2017-04-16:
I am using this library to control my ws2812 ledstrip with an esp8266 nodemcu.
But if a specific color an brightness is set, then after some time; 20 min and sometimes 1 hour, some leds of the ledstrip stop giving light. Mostly at the beginning of the strip (where data pin is connected to the esp8266). But if i update the light color or brightness all the leds are again shining  to the set color and brightness and the again after some time some leds stop giving light. I am using an resistor at the Din of the led, and a 1000uf capacitor at the 5V source. I am also running other code like arduino OTA and a task for the temperature sensor. What do you think could be wrong? And have you maybe an idea what could be wrong?

I am using the following definitions for the fast led lib:

#define DATA_PIN    6 
#define CHIPSET    WS2812B
#define COLOR_ORDER GRB 
#define NUM_LEDS   

FastLED.addLeds<CHIPSET, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);

And using the following function to update the leds:

void setColor(int inR, int inG, int inB) {
    for(int i = 0; i < NUM_LEDS; i++) {
        leds[i].red   = inR;
        leds[i].green = inG;
        leds[i].blue  = inB;
    }
    FastLED.show();

denledsunky, 2017-05-11:
ok

focalintent, 2017-08-16:
Please ask this on the g+ community - there's a wider range of folks to help troubleshoot what may be a code or wiring or hardware issue.  http://fastled.io/+

closed 2017-08-16 by focalintent