FastLED/FastLED #441: SmartMatrix.h missing

bsenglert, 2017-04-19:
I cannot find the SmartMatrix library anywhere within the FastLed libraries folder. Does it need to be placed in the folder from elsewhere? (such as the SmartMatrix3 library download)

bsenglert, 2017-04-19:
So i grabbed Smartmatrix3 and changed the call to "#include<SmartMatrix3.h>"...Now it says DMAChannel.h is missing. Are there a lot of extra libraries i need to find to make the matrices work?

focalintent, 2017-04-19:
This is more of a question for the smart matrix folks and libraries I'd think - what I have in fastled is just a wrapper around their library which needs to be avaialable.

bsenglert, 2017-04-19:
Yeah, I've asked them too...they said its only for teensy platforms. But I've seen where others are using FASTLED for uno's and mega's...i am struggling with it trying to get several matrices to act as one. Thanks for the reply though.

focalintent, 2017-04-19:
That is correct - fasted only supports the smart matrix on platforms that have the smart matrix library, so yes - the teensy is most likely it. 

Folks using fastled on the unos are using one of the led chipsets that fastled supports directly.  

(Things like smart matrix and octows8211 - where fastled is wrapping around another library - are only available on the platforms where said library is available).

closed 2017-04-19 by focalintent

tochinet, 2022-01-31:
No real need to reopen the thread, but since I have the same question 5 years later, I guess it's worth to explicit the requirements in the example. I'll do a proposal for that.