FastLED/FastLED #442: First LED problems in OctoWS2811 mode w/ Faster or Fastest Optimizations

Swap-File, 2017-04-21:
If Faster or Fastest is selected under compiler optimizations in the Arduino IDE, the first LED in a strip seems to be skipped over when updating a strip.  It will show mostly blank, or occasionally update and get stuck on a color.  The second LED and beyond is still updated fine.

I tested tested with the Arduino IDE 1.8.2 and the latest Teensyduino, with a few versions of FastLED (including the master branch).

I only noticed it because I think that Visual Micro changed it's default to Faster when I updated it (or else I bumped the setting).

Should this be brought up in the OctoWS2811 repo instead, since technically it's not FastLED's problem?  Or is this more of a compiler problem?

Either way, it might be useful to get it documented somewhere.

Sushisource, 2017-05-12:
I have also noticed this - quite annoying.

bechtold, 2017-05-14:
Can confirm the bug.
But "Faster with LTO" and "Fastest with LTO" is working as expected.

Sushisource, 2017-05-19:
The LTO ones still have the bug on my 3.2

focalintent, 2017-08-16:
I think this is an OctoWS2811 bug - as the wrapper just does the bit rotation to get the data in a format OctoWS2811 wants internally in its draw buffer, otherwise OctoWS2811 is doing all the work.

closed 2017-08-16 by focalintent