FastLED/FastLED #444: "# pragma message.." on WS2811 using FastLED 3.1.3

renarsvilnis, 2017-04-23:
Hi,
started using the library on a WS2811 Led strip and noticed that I was unable to use the latest 3.1.3 version. When compiling I get this message: ` #    pragma message "FastLED version 3.001.003"`

Board: `Arduino Nano ATMega168`

Currently fixed it by reverting to previous `3.1` version, as it doesn't throw the error.

```c++
#include "FastLED.h"
#define NUM_LEDS    50
#define LED_PIN     4
#define LED_TYPE    WS2811
#define COLOR_ORDER BRG
#define BRIGHTNESS  255

const int centerColor [3] = {211, 5, 36};
const int edgeColor[3] = {252, 255, 255};
const int minDistanceFromCenter = 0;
const int maxDistanceFromCenter = NUM_LEDS / 4;

int currCenterLed = 0;
boolean forward = true;

CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(BRIGHTNESS);
}

void gradientFromPoint (int centerLed) {
  //  memset(leds, 0, NUM_LEDS);

  for (int i = 0 ; i < NUM_LEDS; i++ ) {
    int distanceFromCenter = constrain(abs(centerLed - i), minDistanceFromCenter, maxDistanceFromCenter);

    leds[i].r = (int) map(distanceFromCenter, minDistanceFromCenter, maxDistanceFromCenter, centerColor[0], edgeColor[0]);
    leds[i].g = (int) map(distanceFromCenter, minDistanceFromCenter, maxDistanceFromCenter, centerColor[1], edgeColor[1]);
    leds[i].b = (int) map(distanceFromCenter, minDistanceFromCenter, maxDistanceFromCenter, centerColor[2], edgeColor[2]);
  }
  
  FastLED.show();
}

void loop() {
  gradientFromPoint(currCenterLed);

  if (currCenterLed > NUM_LEDS) {
    forward = false;
  } else if (currCenterLed < 0) {
    forward = true;
  }

  currCenterLed += forward ? 1 : -1;

  delay(100);
}
```

focalintent, 2017-04-23:
Its a message - not an error - it reports the version of the library you're building with - makes it easier for me to see what version someone is using when looking at a build log.

closed 2017-04-23 by focalintent

renarsvilnis, 2017-04-23:
Aaaa I feel bad now, I was thrown away by the message beeing in red and didn't even noticed that it compiled and uploaded. Sorry for creating an issue! 😞

focalintent, 2017-04-23:
No worries - I need to file a bug w/the arduino folks to _not_ mark pragma message as an error.