FastLED/FastLED #445: CHSV to int?

mgcrea, 2017-04-25:
Not sure if I've missed something in the docs, but I have functions that takes an HTML color as an input (eg. `CRGB::White`) that does seem to be an `int` type.

How can I feed a custom CHSV color to this same function?

```
static CHSV myColor = CHSV(160, 255, 255);
```

Tried `myColor.raw` but got an error.

kriegsman, 2017-04-25:
Hi! For general help questions like this, try the FastLED Google+ discussion group at http://FastLED.io/+
We usually only use GitHub bug tickets for bugs in the library itself. 

Also, this may help:
static CHSV myColor(160,255,255);
Maybe!

kriegsman, 2017-04-25:
Closing this ticket, and we can continue the conversation over on the G+ group.

closed 2017-04-25 by kriegsman