FastLED/FastLED #446: New revision of the WS2812B LED

NetForces, 2017-05-03:
As documented by Adafruit: https://blog.adafruit.com/2017/05/03/psa-the-ws2812b-rgb-led-has-been-revised-will-require-code-tweak/

Apparently the reset time has gone from 50us to 300us. With no way (beside a visual inspection) to know which version of the WS2812B we are connected to, we will need a way to adjust the reset time I guess. Is that value exposed ?

kriegsman, 2017-05-03:
Thanks for the note. We saw that come through and we'll figure out some way to let folks change it as needed.

bwilliam79, 2017-05-29:
Would this cause odd flickering of the LEDs when using fill_rainbow to make an animated rainbow effect? I'm seeing this very behavior with a new strip of WS2812B LEDs I purchased recently, but I don't see the behavior with an old set. Comparing the two, I see my old set is the previous revision and the new set is the newer revision.

closed 2019-01-20 by focalintent