FastLED/FastLED #447: Before the code is still OK, but now noworking!

finalvalue, 2017-05-04:
Arduino IDE 1.8.0 / FastLED 3.1.3 / ESP8266 BOARD -- NODEMCU 

error message: 
Arduino: 1.8.0 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src/Lixie.h:11:0,

                 from C:\Users\ADMINI~1\AppData\Local\Temp\arduino_modified_sketch_540074\NTPClock_ONENET.ino:19:

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003

 #    pragma message "FastLED version 3.001.003"

                     ^

In file included from C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src/Lixie.h:11,

                 from C:\Users\ADMINI~1\AppData\Local\Temp\arduino_modified_sketch_540074\NTPClock_ONENET.ino:19:

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.h:11:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.cpp:7:

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003

 #    pragma message "FastLED version 3.001.003"

                     ^

In file included from C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.h:11,

                 from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.cpp:7:

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:48:0,

                 from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.h:11,

                 from C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.cpp:7:

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<12u>':

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_esp8266.h:21:49:   required from 'class ClocklessController<12, 20, 50, 30, (EOrder)66u, 0, false, 5>'

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/chipsets.h:468:7:   required from 'class WS2812Controller800Khz<12u, (EOrder)66u>'

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:102:52:   required from 'class WS2812B<12u, (EOrder)66u>'

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:294:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 12u; EOrder RGB_ORDER = (EOrder)66u]'

C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.cpp:287:74:   required from here

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<13u>':

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_esp8266.h:21:49:   required from 'class ClocklessController<13, 20, 50, 30, (EOrder)66u, 0, false, 5>'

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/chipsets.h:468:7:   required from 'class WS2812Controller800Khz<13u, (EOrder)66u>'

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:102:52:   required from 'class WS2812B<13u, (EOrder)66u>'

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/FastLED.h:294:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 13u; EOrder RGB_ORDER = (EOrder)66u]'

C:\Users\Administrator\Documents\Arduino\libraries\Lixie\src\Lixie.cpp:289:74:   required from here

C:\Users\Administrator\Documents\Arduino\libraries\FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

what happened

focalintent, 2017-05-04:
You don't mention what version of the arduino esp8266 library you are using or what board your are selecting in the drop down.

focalintent, 2017-05-04:
Ah - I see the problem there is no pin13 for the nodemcu layout - if you are trying for raw PIN numbers vs what's on the board then you should add

```
#define FASTLED_ESP8266_RAW_PIN_ORDER
```

Before you include FastLED.h

closed 2017-05-04 by focalintent

finalvalue, 2017-05-04:
thank you for your anwser!  i change the select . it's OK now.