FastLED/FastLED #449: ws2813 new version with different timing

redPanther, 2017-05-08:
Hi,
I had ws2813 running smoothly for a while with driver implemented in current fastled. Now I have another ws2813 and it doesn't work - it flickers all the time. I found out it works with 2812b driver, but I think this is only a lucky coincidence.
I got the new datasheet for the stripe and - oh suprise - the timing is changed.

old:
[image]

new: (revision april 2016)
[image]

datasheet:
[WS2813 IC LED information.pdf](https://github.com/FastLED/FastLED/files/983089/WS2813.IC.LED.information.pdf)

can you also support the new version of ws2813.
thx!

artag, 2018-04-01:
There now seems to be another revision of the timings.

http://www.world-semi.com/DownLoadFile/136

focalintent, 2018-04-02:
This is annoying - one of the things i'm doing to the clockless drivers as part of the RGBW/16-bit re-write is making it easier to define timings based on what's in the datasheets, so people can quickly/easily try different timings, for cases like this when datasheets/timings change...

artag, 2018-04-02:
I'm not sure how you'd select the appropriate timings other than experimentally. 
I started looking into this because a set of WS2813 leds running from pjrc's Octows2811 code were flickering sometimes. So far, the only explanation I've found is that the timings are at the limit of some specs and the LEDs were cold.

focalintent, 2018-04-02:
Eventually, as part of the addLeds specification, people will be able to adjust their timings e.g. be able to say something like:

```
// define timings in nanoseconds - total per bit, T0H, T0L, T1H, T1L, reset
class my_WS2813_timings : ClocklessController<1100,350,750,750,350,300000>;
...
void setup() { 
  FastLED.addLeds<my_WS2813_timings, DATA_PIN, BGR>(leds, NUM_LEDS);
}
...
```

I don't have the specific details worked out, but it's become clear to me over the past couple years that I need to make it easier for people to try alternate timings.

artag, 2018-04-02:
The Octows2811 code allows changing of the 'high' length of 1 and 0 bits (and the reset time), which doesn't seem enough to meet the specs but probably is. I wonder what difference the 'low' time makes, when the high is defined and the cycle time is also apparently fixed.

The leds I had trouble with produced regenerated bit times of   330 and 650 ns. It could be interesting to see if that has a temperature dependence.

focalintent, 2018-04-02:
It might be enough to just define the total and high times, and let the low times be determined automatically working backwards from that.  The main reason I'd consider also having the flexibility for defining both high and low times for both 1 and 0 bits is I have seen data sheets where the timings for 1 and the timings for 0 don't add up : )  But you're right, there'd probably also be a simpler version that simply defined bit time, high times for 1 and 0, and reset time.

focalintent, 2018-04-02:
I've also found that the timings that the leds worked with could be "tweaked" by adjusting the voltage going into the leds ... it all ends up being very hand wavey, and makes me wish clock+data controllers/chips were more common!

artag, 2018-04-02:
You can be sure setup and hold specs would be just as vague !

I started using the 2813s because I had a lot of single-led failures in a batch of 2812s with the resulting loss of the rest of the strip. But I'm unsure how much they help. I'm sure they don't test the leds individually, so I wonder how many leds in a strip are already using their 'fallback' data when shipped ?

artag, 2018-04-02:
Another interesting possibility is to receive the data from the far end of the strip and check for corruption.

focalintent, 2018-04-02:
Writing avr code to catch the data at the other end is ... tricky :)

artag, 2018-04-02:
Yes :)
Could do it for a specific test (wait for rising edge, read repeatedly, analyse results)  but not as a general error check. Maybe the Uart could be tricked into doing it.

focalintent, 2018-04-02:
The trick would be getting the timings to line up right - I don't think you'd want to do it on something running at only 8Mhz -- you'd probably want something running at a higher rate that gave you 
a  prayer of telling you what your output timing was looking like as well...

artag, 2018-04-02:
Yes, you want to look at the regenerated lengths and choose something halfway between

closed 2019-01-20 by focalintent