FastLED/FastLED #453: Compilation error using an example.

yo-shari, 2017-05-24:
Hello, I'm new to using this library. I was that there is a commit message that operator= was fixed in pixelset.h however when I am trying to compile the blink example I get an error with that exact line.

[image]

focalintent, 2017-05-24:
Have you gotten the most recent version of the library from GitHub?

yo-shari, 2017-05-24:
Yes, I downloaded the library earlier today. Then again just now.

focalintent, 2017-05-24:
What happens if you try with a more recent version of Arduino?  This is the first time i've seen this error pop up - and I stopped doing work/testing with the 1.0.x versions of arduino a year or more ago - the readme even states this:

IMPORTANT NOTE: For AVR based systems, avr-gcc 4.8.x is supported and tested. This means Arduino 1.6.5 and later.

yo-shari, 2017-05-24:
I was using a mac and I got it working using a PC

marmilicious, 2017-05-25:
What version of Arduino is on the PC?

yo-shari, 2017-05-25:
The PC must have a new version because it wasn't the right version on the mac. My bad, thank you for  helping me out though. I got the LEDs working!

closed 2017-08-16 by focalintent