FastLED/FastLED #454: FastLED with Arduino Genuino 101

alkagarg, 2017-05-27:
I am not able to compile my fast LED code to my Arduino Board. Do you know how I can fix this problem?

My fastLED library version is 3.1.3 

This is the Error:
In file included from /Users/aayush/Documents/Arduino/libraries/FastLED/led_sysdefs.h:27:0,
                 from /Users/aayush/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /Users/aayush/Documents/Arduino/libraries/FastLED/bitswap.h:4,
                 from /Users/aayush/Desktop/LED/LED.ino:1:
/Users/aayush/Documents/Arduino/libraries/FastLED/platforms/avr/led_sysdefs_avr.h:12:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
                    ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino 101.

jasoncoon, 2017-05-27:
Looks like the [Arduino 101](https://www.arduino.cc/en/Main/ArduinoBoard101) uses an Intel Curie microcontroller, which isn't supported by FastLED.

focalintent, 2017-08-16:
And it is unlikely to ever support it, as intel has discontinued the curie.

closed 2017-08-16 by focalintent