FastLED/FastLED #455: Getting errror with NodeMCU DevKit 1.0 after a while

pipoblak, 2017-05-29:
Im getting this error after a while using FASTLEDS in my nodemcu.

the code:
https://gist.github.com/pipoblak/344c8b93625dff744fd296a73a78da38
the stack trace:

```
Connecting to QG BOLADAO
......
WiFi connected
IP address: 
192.168.1.76
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
mac 981

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
þ
```

pipoblak, 2017-05-30:
I tryed a more simple code with FastLeds and NodeMCU and dont work too.

Code: https://gist.github.com/pipoblak/9e164e86242728e14f72500d2de9eeb4

StackTrace:
```
dev 1143

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
á

```

jasoncoon, 2017-05-30:
Do the FastLED examples work?

pipoblak, 2017-05-30:
I will test examples with NodeMcu , but that code run without problem with Teensy,

closed 2019-01-20 by focalintent