FastLED/FastLED #457: Flickering leds (esp8266)

eldoner, 2017-05-30:
I have a problem, with leds. Examples from library does not work properly, sometimes strip is only partialy lighted, and of course a lot of flickering. I use [Nodemcu lolin v3 (esp8266)](https://img3.banggood.com/thumb/view/oaupload/banggood/images/6B/97/3cda1ef7-adef-92d8-f26b-9c047864b814.jpg). Everything is wired like [there](https://cdn-learn.adafruit.com/assets/assets/000/030/892/medium800/leds_Wiring-Diagram.png?1456961114). There is common ground, resistor and capacitor. I use TXS0108E level converter to step up data line. I tried almost all pins on Nodemcu - no diffrence.  Led type: WS2812B.

On the [movie](http://sendvid.com/m5adtx43) I use example ColorPalette, where I added:
```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 
#define FASTLED_ALLOW_INTERRUPTS 0
```

MFornander, 2017-05-30:
See the FAQ and ask on the G+ group, not here please.  GitHub is for code issues and since examples have the same issue on your setup, it's not a code issue.  That said, I had trouble with cheaper level shifters from Ebay that went away with a more traditional 74HC245.

The message is normal since 8266 doesn't drive LEDs using SPI.

erhan-, 2017-06-22:
Same problem in similar setup but I am using SN 74AGCT125 N

debsahu, 2017-11-14:
Add these too:

```
#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
#define FASTLED_INTERNAL
```

Smitty357, 2018-09-23:
> Check the voltage of your power supply. Maybe it's too much.
Which voltage level are you referring to?
Will the additions below help?

> #define FASTLED_ESP8266_NODEMCU_PIN_ORDER
> #define FASTLED_INTERNAL

closed 2019-01-20 by focalintent